betaflight/betaflight #3214: Omnibus F7 - no Level Mode (Horizon/Angle)

RipperDrone, 2017-06-05:
Switch is set correctly in BFC, still Ominus F7 does not engage into Level mode. Can only fly acro / Airmode so far...

If all Omnibus F7 related issues should be subsummized in one issue report, pls let me know (there is at least one other F7 Omnibus related issue so far).

RipperDrone, 2017-06-05:
Seems to be defective on one quad (nightly build), but ok on another Omnibus F7. Maybe there have been updates preparing the gyro selector (CLI switch set gyro_to_use = 0?) so Self-Level Mode is currently corrupted?

nathantsoi, 2017-06-05:
works fine for me did you calibrate your gyro and acc?

RipperDrone, 2017-06-05:
found that copy-paste of CLI yielded some bad values for PIDs of LEVEL, particularly I=0! Easy fix, but hard to find. Seems like latest BF3.2 nightly with latest ('nicghtly') BFC 3.2 vs. older 3.2 versions have backwards compatibility issues...?

RipperDrone, 2017-06-05:
I was too quick: Defaults are I=0 for VEL. Horizon has strength = 20, angle limit = 55, sensitivity = 55. 

When testing AUX inputs in BFC, they are all good. Still quad does not fly level mode... :-(

RipperDrone, 2017-06-05:
@nathantsoi yes, did that. And gyro + acc sensor readings look perfectly ok in BFC

RipperDrone, 2017-06-05:
Tried to re-flash, set to defaults, then manually set each value from diff of other copter. Now working again.

Not sure what was going on here... will monitor closely...

closed 2017-06-05 by RipperDrone

mikeller, 2017-06-05:
@RipperDrone: You sometimes get these things when working with nightly builds, since the configuration format changes, but the firmware doesn't detect a change in version number, so the old format configuration is misinterpreted. Usually a 'Reset to defaults' / 'Full chip erase' will fix it.