betaflight/betaflight #3216: Omnibus F7 - Self-Level / Horizon Mode not working

RipperDrone, 2017-06-05:
Just did a couple more hover tests outside, and it seems I have to reopen this issue (old thread was confusing so here is a new one):

Horizon Switch is set correctly in BFC and gyro is calibrated, still Omnibus F7 does not engage into Level mode. Can only fly acro / Airmode so far...

It seems to behave differently in Horizon Mode regarding control inputs, but it's still in Acro. Maybe it has to do with the PID settings? I have double checked them all, and they seem to be at their defaults.

[image]

If all Omnibus F7 related issues should be subsummized in one issue report, pls let me know (there is at least one other F7 Omnibus related issue so far).

RipperDrone, 2017-06-05:
Shouldn't the Horizon Transition value default be set to 100 instead of 0?

RipperDrone, 2017-06-05:
Bug identified! Default Transition value was incorrect - should be set to 100 instead of 0 (zero means that at zero throttle, there is a transition of attitude from self-level to acro mode, correct?)

Set it to 100 and Horizon mode now works like a charm. Since I'm not familiar with creating Pull Requests: Can somebody adjust the default values accordingly?

RipperDrone, 2017-06-05:
This is the initialization code in pid.c It seems that the values are not transferred correctly into BFC (nightly)?

[image]

mikeller, 2017-06-05:
Have you tried 'Reset to defaults' / 'Full chip erase' to force the defaults to be loaded?

RipperDrone, 2017-06-05:
Shame on me, I had marked out the defaults CLI command (to run the whole diff without rebooting the FC). Erase I never do with full erase, so probably retained some old values when flashing the upgrade.

Tried defaults now - this loads all values into BFC edit fields as it should. So definitely it's just a migration problem... closing this issue.

closed 2017-06-05 by RipperDrone

mikeller, 2017-06-05:
All good. When using released versions, the way you did it should work fine, since the configuration version change is detected and caused a configuration reset. When working with latest, this doesn't apply, and this trips me up occasionally as well.