betaflight/betaflight #3217: The "level_limit" parameter has no real effect in Angle Mode

MJ666, 2017-06-05:
The default setting is 55 for this parameter. This should mean a maximum of 55 degrees angle. In the CLI you can change this from 10 to 120. 120 degrees may even not make much sense. Wit 90 degrees the copter will be completely vertical and can not maintain height even with the most powerful motors. Anyhow in real flight changing this setting will not have any real effect? The possible angle stays more or less the same and is very low. Any thoughts on this?

MJ666, 2017-06-28:
Nobody has an thought here?

borisbstyle, 2017-06-28:
Angle limit is constraining the max angle
https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c#L344

What do you mean with no effect? That the parameter doesn't work for you?

MJ666, 2017-06-28:
The change of level_limit in the CLI from 55 to i.e. 90 will not show any effect during flying. The resulting angle during flight locks to be more or less constant at an relative low value.

MJ666, 2017-08-28:
No one has an idea here? I know most are ACRO mode pilots here. Anyhow I think it should be fixed. IT was definitely working in older versions of Betaflight. I was looking into this part of the code but did not find the root of the problem.

takeshisama, 2017-08-29:
Funnily enough, someone linked this post on [reddit](https://www.reddit.com/r/Multicopter/comments/6wutxz/any_smart_software_developers_here_please_help_me/) so I messed around with it a bit and it looks like the level_sensitivity needs to be increased to match the max angle.  At least that worked in my tests, hopefully it helps either resolve or clarify the intended uses of the settings.

MJ666, 2017-08-30:
Looks to be we have here two parameters effecting the same thing. This is a bit redundant to me? Was this really intended?

martinbudden, 2017-08-30:
@MJ666 , it looks like `levelAngleLimit` was intended to set the limit, and `levelSensitivity` was intended to change how quickly the aircraft reached the limit according to the stick position, but the logic was flawed. I think the simplest thing is to remove `levelSensitivity`. I've done this in PR https://github.com/betaflight/betaflight/pull/4013 .

Can you test this and confirm it works for you?

MJ666, 2017-08-31:
Will test next days. Many thanks.

MJ666, 2017-09-03:
Tests go quite well and #4013 solves the issue.

closed 2017-09-03 by martinbudden