betaflight/betaflight #3218: Sparky2 Accel Not Recognised

ukmook, 2017-06-05:
On latest Release stable release 3.1.7 accel is not picked up on configurator, tried an earlier firmware with same result. Tried a much earlier release 3.0.1 with success. 
Looked back through code and noticed a difference  in 

target.h
 #define EXTI_CALLBACK_HANDLER_COUNT 1 // MPU data ready

was omitted, forked added the line, built the firmware reported as 3.2.0 and all is well. 

I am not a coder, so just to be sure i have checked the board with INAV and dronin which both report good accel.

mikeller, 2017-06-05:
Good finding, @ukmook. Would you be able to open a pull request for this change?

ledvinap, 2017-06-05:
That's strange.. EXTI_CALLBACK_HANDLER_COUNT should be superceded by exti io

mikeller, 2017-06-06:
@ledvinap: Indeed. Also, EXTI_CALLBACK_HANDLER_COUNT was surprisingly alive for something that should be dead. See #3221.

ukmook, 2017-06-06:
@mikeller do you still want me to do a pull request as i see you have been working on this issue? 
Just got back from work, apologise for the late reply.

mikeller, 2017-06-06:
@ukmook: No, sorry, @ledvinap is correct, and EXTI_CALLBACK_HANDLER_COUNT is not used any more. But it would be great if you could do some further testing to isolate a fix for the SPARKY2.

ukmook, 2017-06-07:
@mikeller I am finding that sparky2 wont always connect to GCS, i know @digitalentity did some work to get the mpu9250 working with inav and have found a  PR [https://github.com/iNavFlight/inav/pull/941](url)
This may or may not fix the issue.  I'll talk with him tonight also i have a new sparky2 which i will try tonight.

martinbudden, 2017-06-07:
@ukmook , have you tried the latest 3.2 build?

ukmook, 2017-06-07:
@martinbudden i am using 3.2.0 that i built using the fix above i can build from scratch and try that.

ukmook, 2017-06-07:
@martinbudden Tried 3.2 the board will not connect to the gcs, driver drops out reporting unrecognised device, thought it might be my system so tried a few things even linux but no joy. 
Reverted the Firmware back to 3.0.1 and its working Ok no connection issues and the accel and gyro both working. INAV and dRonin both work Ok. Did this to prove its not a hardware issue.

ukmook, 2017-06-07:
All working, corrected a pin error on vbat sense, removed soft serial, changed serial port count to 4, changed the serialRX_uart to correct port number (6), connects with no issues and reports all sensors in GCS.
GCS Reports Version # Betaflight / SPARKY2 3.2.0 Jun  7 2017 / 17:35:23 (d68a72d)

mikeller, 2017-06-08:
@ukmook: Can you please submit a pull request for your fixes?

closed 2018-04-21 by DieHertz