betaflight/betaflight #3219: BF F3 board loses yaw after arming

techcoast, 2017-06-06:
I'm not sure this is the correct forum to post this in, but here goes. I recently pruchased the BetaflightF3 board and mounted it. Everything seemed to check out in Betaflight, ESC's calibrated, and Rx showed full control, etc.  But, when I took it out to maiden, as soon as I armed the quad started to rise with throttle at the bottom and I had no yaw control. I repeated this process a few times and each time with the same result. I came home, plugged it all in and (without the props) I can arm the quad have no yaw control, but still have the ability to disarm and the pitch and roll work fine. I can also throttle up and down. (Assuming my min throttle was too high as to let it take off from the git go-- unlikely but I'll look into it) Help?!

RipperDrone, 2017-06-06:
Check Yaw P in BFC, sometimes when upgrading firmware or configurator values get confused. Mine was set to zero at one point.

fftunes, 2017-06-06:
Did you arm with sticks? If so, then there's no yaw control below min_check i think. Use switch & arm mode instead.

Did you calibrate escs?

techcoast, 2017-06-06:
The values were 70 and 100.. so no 0 value. I do arm with sticks, and still no yaw control. I have arm on yaw on all my models.. so I'm not really keen to start a new thing at this point. I did calibrate the ESC's. At this point I have replaced the FC with a new one, so let's see how this goes.

techcoast, 2017-06-07:
Ok, so new FC same issue. Here's the option. Put the arm on a different channel. I really hope BF fixes this since all my models are on yaw arm and I'd hate to have an accident because I forget to use a switch.

mikeller, 2017-06-07:
I don't know what you want Betaflight to fix. As @fftunes remarked, this is the expected behaviour. If you want to use stick arming, this requires yaw control to be off below `min_check`, or the craft would flip on every arm / disarm.

techcoast, 2017-06-07:
I did put my min_check to 1050 which is 10 below my min throttle at 1060 per advice from one very knowledgeable person. Again, arming isn't the issue. The copter arms just fine, but when I arm, I lose control over yaw. I can still disarm via yaw movement, but can't control yaw in flight at all. I don't think that's normal lol.

mikeller, 2017-06-08:
@techcoast: Can you check your setting of `yaw_motor_direction` and `yaw_control_direction`?

techcoast, 2017-06-08:
Y-motor = 1
Y-control = 1

fftunes, 2017-06-08:
The issue with the quad ascending after arming with zero throttle is still the same? This might be a sign of excessive (physical or electrical) noise, unrelated to yaw not responding at all.

Also, if your throttle channel goes down to 1000, you could set min_check to something like 1015-1020.

What type of rx do you use? Something serial or PWM (one wire per channel)?

techcoast, 2017-06-08:
No, it seems to stay on the ground, I have throttle control. I just don't have yaw after arming. 
My channels do hit those values. I'm ausing an XM+ FrSky rx with SBUS. Even with min_check at 1015 yaw control is still not available after arming.

mikeller, 2017-06-08:
@techcoast: So when you fly it, it does not gain yaw control after it is airborne?

techcoast, 2017-06-08:
Correct, once the motors are armed, yaw control is gone.

techcoast, 2017-06-20:
Anyone have any ideas?

Noctaro, 2017-06-20:
I dont know if it can help. I had this happening when i switched to a untouched pid profile on my brushed build. I did this by mistake using my osd and was not aware of it. At liftoff i had control of all axis except from yaw. Did not try to reproduce for now.

diabolicaldan, 2017-09-06:
I have the same problem recently after crash. I plug my FC back into betaflight and took a look at how the gyros were responding in the Setup Tab. All gyros were reporting back normally with the exception of yaw. No matter what I did when I spun the quad copter on its  yaw axis it would not show me any feedback in the picture displayed in the setup tab.  Leads me to believe that the Gyro took a pretty serious hit and now it's pooched.  Not sure if you have the same problem... but I am curious is you can see response from your yaw gyro sensor.

martinbudden, 2017-09-06:
@techcoast , can you post a blackbox log?

closed 2018-04-21 by DieHertz