betaflight/betaflight #3220: Have an old Flip32 board, unsure what firmware to flash to it!

Pukington, 2017-06-06:
Hey guys, I'm sorry if this is out of place.

I have a half built ZMR250 based quad from around early 2015 that i had to box without finishing :(work commitments).  Now i've some time i want to play with it and heard betaflight is all the rage.

So i think i've a flip32+ board, i remember it being a naze32 clone and want to flash betaflight, it's currently on a very old cleanflight build (i think).

I'm a little confused as to which firmware i should flash, since it's a naze clone, should i use the betaflight_3.1.6_NAZE.hex firmware?

Thanks for any input!

jflyper, 2017-06-06:
@Pukington If it's a NAZE clone/architecture, you should use NAZE hex.

Please note that GitHub is for questions and discussions around the development of the Betaflight firmware. For help with your setup, please search the RCGroups first. Betaflight discussion thread will be the nice choice.
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

Pukington, 2017-06-07:
@jflyper Thanks for the reply mate, i'm sorry this is an inappropriate place.  I appreciate your help and shall head over to RCG for further concerns.

Thanks again!

closed 2017-06-07 by Pukington