betaflight/betaflight #3222: Smart

yujanshrestha, 2017-06-06:


hydra, 2017-06-06:
I don't think we want even more cli commands.  Most people don't have a laptop out in the field anyway.  There are stick combinations to change vtx channels/bands, does this not work for you?

IB1387, 2017-06-06:
I don't think changing SmartAudio over the CLI is a comfortable way to change vtx channels / transmit power. I would rather see vtx channel/band/transmit power adjustments over stick combinations or similar to inflight adjustments.

mikeller, 2017-06-06:
Couple of points:
- it already is possible to control the TBS Unify via CLI: Since the VTX settings are stored as part of the flight controller config, simply change the settings with `set ...`. The flight controller will reboot when you exit the CLI, thus activating the stored settings;
- there is work under way to make LUA scripting available to users of ersky9x: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/30 ;
- I agree with @hydra re the value of this use case. As an alternative, channel stepping could be implemented as an adjustment value (i.e. use a momentary switch on your TX to cycle through the channels.

yujanshrestha, 2017-06-06:
Thanks for the pointers folks. The main problem I have with stick combinations is how can you be sure what band / channel you are actually on? I mean I guess you could assign two 3 position switches but that would just give you 9 options. Without telemetry or osd it is impossible short of just looking at the vtx itself, which is difficult if not impossible for certain builds. You may just end up listening in on a neighboring channel and just show static after a few meters. However with a cli command there is no question what frequency  or transmit power you are on. While I agree this use case will not be widely used I figured it would be trivial to expose such a feature so the developer time vs usage frequeny equation balances out.

@mikeller 
You have my hopes high with the `set ...` command. This would be perfect! However, I am unable to find the correct setting. I did a cli dump and combed through the code a bit but can't find anything to set. Of course, I am unfamiliar with the codebase so maybe I am missing something.

hydra, 2017-06-06:
@mikeller and @yujanshrestha I added some thoughts on exactly how to do this here:

https://github.com/betaflight/betaflight/issues/2392#issuecomment-282563944

yujanshrestha, 2017-06-07:
@hydra thanks for referencing that discussion. I am looking forward to seeing that get merged. Looks like the CLI / config var approach is an important part of that architecture and would entirely satisfy my needs.

I am pretty comfortable using pretty low-level commands. Is there something like a remote procedure call where I can call a C function from the CLI. Then maybe I can just call the same function the osd uses to interact with smart audio but do it from the CLI without an osd at all. Shoot, I am even willing to just issue raw UART commands if someone gave me a pointer on how to do so.

I can imagine a new use case similar to the TNR Wand. It can be a mobile app that can be used to remotely set the vtx channel / transmit power by just plugging it into the usb using an otg cable. I would imagine this can just use a CLI command as well. This can be done even before plugging the battery in and on power up the FC can just issue the proper commands to change the channel to the one stored in config memory. No more scrolling through channels and shooting other flyers out of the sky. Or even a tab in the betaflight chrome app similar to the IIRC vortex chrome app that lets you choose vtx channel / power.

jflyper, 2017-06-07:
@yujanshrestha Features you want has been implemented as a part of the vtx code consolidation phase 2, but it was stashed due to conflict with some other changes. I'm hoping that I can revive the work soon.

yujanshrestha, 2017-06-07:
@jflyper Thanks bud. I know a good software design when I see one. I like your design sketch, its got abstractions in the right places. I hope you can resume work on this as I can see lots of use cases for a nice cli api. In fact I would be willing to give you a donation if that helps speed up the process :).

ethomas997, 2017-07-15:
So, is the plan to have the vtx_band/vtx_channel/vtx_power settings work with TBS SmartAudio?  How would the SmartAudio-VTX option be enabled?  Is anyone working on this aspect of it currently?  I'd really like to see it implemented for SmartAudio VTX.

--ET

jflyper, 2017-07-17:
@ethomas997 SmartAudio is still enabled by ports tab config. Band, channel and power are the minimal set that will be supported.

ethomas997, 2017-09-03:
I've posted a PR on Cleanflight that implements the CLI commands for SmartAudio and Tramp.  https://github.com/cleanflight/cleanflight/pull/2936

--ET

yujanshrestha, 2018-01-31:
:) Great! closed by https://github.com/betaflight/betaflight/pull/4265

closed 2018-01-31 by yujanshrestha