betaflight/betaflight #3223: Smart-port telemetry work only for a few seconds

siamect, 2017-06-06:
I have Omnibus F4 Pro V2 with sd card and BF 3.1.7
 Smart-port telemetry work only for a few seconds after restart and after saving anything from the configurator. After that the updates of the sensors stops in the Taranis except RSSI and RxBt.  I get sometime "sensor lost" voice messages.
I use XSR receiver with the inverter hack. 
I have the luascript on the Taranis to set the pis rates and vtx parameters via smart audio. I can get this to work only if I save something from the configurator. When I do this the parameters are coming up and I can change them and save them.

I also have a CC3D revo F4 with BF3.1.7 and on that one it works fine...

Arakon, 2017-06-06:
There was a XSR firmware released a day or two ago that fixes a smartport/telemetry issue. Give it a try.

RipperDrone, 2017-06-06:
RX power is stable? Have the same config working on my Omnibus F4 v3 (non-pro), no probs. VTX powered direct from LiPo, cam powered from VTX.

siamect, 2017-06-07:
Thanks for the tips.  
@RipperDrone. Your comment is very valid because the Omnibus F4 Pro V2 is sadly having serious design flaws that makes it less suitable for supplying 5V to anything. The 3A BEC is ok but it supplies the 5V through some diodes that brings the voltage down to whatever level it feels like and with only the XSR connected it is below the 4.75... therefore they have cranked up the voltage from the BEC quite a bit but that's not a good solution...    
So I have fixed that problem with a work around for now (don't recommend that workaround, because it still goes through one diode so I will not describe it unless someone ask specifically).  
So yes my voltage is stable on about 4.95V and there is not much ripple either. 
 
@Arakon I think that can be the solution. My other drone has a X4R-SB receiver and that one works...  
 
Big thanks to both of you!

siamect, 2017-06-07:
Yes I flashed the XSR with the new firmware and telemetry works.   
So far so good but now I cannot get any values into the luascript menus on the Taranis to set the pids rates and vtx parameters via smart audio.

yujanshrestha, 2017-06-07:
I had the exact same issue. Telemetry for a second -> Flash XSR -> Telemetry works but no pids. The problem was the XSR itself. I tried swapping it with a X4RSB and I started getting pid rates. But still no vtx params.

I am linking this [RCG discussion here](https://www.rcgroups.com/forums/showthread.php?2793214-Modify-Betaflight-PID-VTX-Settings-w-TARANIS-LUA-Script-Menu/page16)

siamect, 2017-06-07:
@yujanshrestha  
Thanks for this info, at least I'm not the only one...

siamect, 2017-06-08:
ok so I just flashed the XSR_nonEU_build151118 which is the old version and it works now, well based on 5 minutes testing :)   
So the question is what version was on there when it was delivered?

closed 2017-06-08 by siamect

yujanshrestha, 2017-06-08:
By any chance @siamect are you able to get the lua script working? See the pid values?

siamect, 2017-06-09:
@yujanshrestha  Yes, when I flashed the XSR_nonEU_build151118 then the luascript started working so I can see and change the pids, rates and the vtx settings.

piepluquad, 2017-08-18:
i had same issue, using softserial for smartport telemetry : i had to lower baud rates on the softserial port used AND i had to set both softserial1 & softserial 2 to the same baud rate (even if softserial 2 was not in use)

jflyper, 2017-08-19:
@piepluquad 
> i had to lower baud rates on the softserial port used AND i had to set both softserial1 & softserial 2 to the same baud rate (even if softserial 2 was not in use) .

S.Port is running at 57.6Kbps fixed and changing the speed does not affect how serial ports talk to the receiver. Also, while software serial ports do have constraints on how they could be configured depending on the pins/timers used for them (say they are linked for such configuration), changing speed on unused serial port does not affect the behavior of linked ports.

ignaciobfp, 2017-09-02:
I am experiencing this same problem, with Omnibus F7 V2 and a hardware UART for telemetry. In my case it is not only PIDs, all telemetry is gone as soon as about 5 seconds pass since power on. I will try to upgrade/downgrade firmware on XSR as suggested.

asg-8, 2017-09-09:
@ignaciobfp Thanks!
I swapped an X8R (with telemetry and PIDs/rates working) with a new XSR a few days ago. I had the same problem, no telemetry nor PIDs/rates (Lua Script).
I have just updated my new XSR with XSRLBT170619.frk (EU version) using the Taranis X9D Plus and telemetry inlcuding PIDs/rates works perfectly!