betaflight/betaflight #3224: SBUS FrSky Betaflight / LUXV2_RACE 3.2.0

adrianmiriuta, 2017-06-06:
SBUS Frsky is not working on UART4
saving not possible 
after reboot in configurator 
Receiver "select one"
working backup config from 2.1.7

mikeller, 2017-06-06:
I suspect that you have some other function selected for UART4 as well. The firmware will detect if an invalid combination (most of them are) is selected on a port, and remove all functions from that port.

Please note that GitHub is for questions and discussions around the development of the Betaflight firmware. For help with your setup, please search the RCGroups first. Betaflight discussion thread will be the nice choice.
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

adrianmiriuta, 2017-06-06:
@mikeller 
the port is not used , serialRx is activated on ports tab
the settings work with release fw 3.1.7
they do not work with git master fw 3.2.0
setup is ok  it is a firmware issue!

mikeller, 2017-06-06:
Can you post a CLI `diff` please?

adrianmiriuta, 2017-06-06:
@mikeller 
[cli_diff_v10.txt](https://github.com/betaflight/betaflight/files/1056315/cli_diff_v10.txt)

mikeller, 2017-06-06:
```
serial 3 64 115200 57600 0 115200
```

Looks like you've got serialRx already defined for UART3.

adrianmiriuta, 2017-06-07:
serial 3 
is frsky sport telemetry TX  (telemetry is working)

this is a restore on 3.2.0 fw 
of a working config dump made in 3.1.7 fw
without any changes.

adrianmiriuta, 2017-06-07:
Closed!
3.1.7 dump incompatible with 3.2.0
looks like restoring a 3.1.7 dump on 3.2.0 fw messes up serial ports
manually setting UART4 as "Serial RX" after "Full chip erase" works

closed 2017-06-07 by adrianmiriuta

mikeller, 2017-06-08:
Yes, this is a case where a manual update is required after the update. Since the UART definition for this target was changed, UART numbers in the dump changed as well.

slash4x4, 2017-11-15:
I have a lux V2 flight controller updated to 3.2.2 and no longer get Telemetry or vfas readings. was working fine on 3.1.6. xsr RX is on uart 4 s.port wire is gooing to uart5 . What do I do ???? Please help