betaflight/betaflight #3225: blheli Pass through not working F7 Anytime

mattyra, 2017-06-06:
See the log below, the blheli suit is not running correctly some times it finds 1 esc some time 3 esc's 

Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.14
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 74, Baud: 115200
Serial port successfully opened with ID: 74
MultiWii API version received - 1.35.0
Flight controller info, identifier: CLFL, version: 2.0.2
Running firmware released on: Apr 22 2017 14:22:23
Board: ANY7, version: 0
Unique device ID received - 0x3500323335511735303934
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 3 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 4 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
Reading setup finished
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 3 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 4 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
Reading setup finished
Reading setup
[... 659 lines cut ...]
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 3 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,5,255,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup finished

DieHertz, 2017-06-10:
We really need to debug these passthrough issues appearing after 3.1.x release.
Unfortunately BLHeli Configurator logs are of no help to it, the issue appears to be happening between the FC and the ESCs, BLHeli Configurator just gets a "can't communicate with ESC" report from the FC.

brucesdad13, 2017-06-10:
I used it yesterday to configure 6 EMAX Bullet 12A ESC for a hexa x.

RipperDrone, 2017-06-12:
I bricked one ESC due to flashing freeze. Omnibus F7, BLHeli_S ESC.

rabe-, 2017-06-12:
i've had problems on my F4 board too. the chrome app configurator worked, but flashing or connecting with the blheli suite did not.  using the CLI to passthrough a single ESC at a time worked though, so i flashed one at a time.

closed 2018-04-21 by DieHertz