betaflight/betaflight #3228: Change RGB LED color based on VTX channel (Feature request)

oxivanisher, 2017-06-07:
It would be a very nice feature, to have the RGB "base" color for the LEDs on the copter configurable, so that it changes depending on the VTX channel from SmartAudio and the TrampTelemetry. Since TBS is making their Race Tracker more versatile (and others will probably follow), it would be so much easier to have a fancy colored LED race if those colors would match up more or less automatic.
Since I do not know if the RaceTracker then has configurable colors for each channel, or if the colors will be fixed for the channels, I can not say how the implementation should look like at the moment. Also this feature is so brand new, that the new RaceTracker firmware is not out yet to have a look at it. But since that will happen within the next days (1-3 corresponding to trappy), I would gladly update this issue with more detailed information once it is released.

For more information about this feature and what it all is about, watch this demonstration from trappy during the FPV Life stream: https://youtu.be/B9gdAIKpmAs?t=1h32m35s

sheaivey, 2017-06-09:
This feature would be great and is going to be implemented on the LaForge VTX accessory port. But we
Would much rather have the LED attached to the FC.

TBS has already released their color scheme along with the SmartAudio protocol.

``` 
/* FROM TBS SMARTAUDIO PDF */
H = (current freq – 5645) * 1.2
S= 1
V = 255
HSV Ref:  https://en.wikipedia.org/wiki/HSL_and_HSV
```

I could see this feature being implemented as another LED mode like the blinker, watning, blink always or throttle etc. The blink mode could be enabeld if the used has smart audio active.

I'll take a look at the code base and see if there are other tickets around this issue. If it is not already being worked on I could start on it.

sheaivey, 2017-06-09:
Looking into this a little further this could be implemented pretty easily by adding another overlay type such as LED_OVERLAY_VTX.

With the LaForge VTX we display a little bit more information with the LED Strip.
- Pit Mode - when pit mode is active the LEDs are dimmed.
- SmartAudio - Any time the settings are changed we blink with the updated settings for 3 seconds.

Displaying settings on the LED looks something like this and only displayed for 3 seconds then reverts back to frequency color.
```
LED 1: Green - Blinks when SmartAudio settings are updated.
LED 2: Orange - Power set to >= 25mW or OFF
LED 3: Orange - Power set to >= 200mW  or OFF
LED 4: Orange - Power set to >= 500mW or OFF
LED 5: Orange - Power set to >= 800mW or OFF
```
If LEDs 2 through 5 are off this indicates it is currently in Pit Mode.

I'm going to take a shot at adding the overlay this weekend since I have already implemented on the LaForge VTX.

aukrausmi, 2017-06-10:
definitively makes sense to align a feature that would work cross platform and bring some unity to the scene FPV

bilson, 2017-06-10:
try with https://betaflightgroup.slack.com/

sheaivey, 2017-06-10:
OK, I have this working in my fork. I wound up having to replace an existing overlay because there was no more bits left for a new overlay. The overlay I replaced was BLINK_ON_LANDING (not sure this will be missed). If we want to implement this a different way I'm up for suggestions however this was the least intrusive way without having to check firmware version in the configurator.
I also updated the configurator to reflect the new VTX overlay.

I was able to get this working as described [here](https://github.com/betaflight/betaflight/issues/3228#issuecomment-307451346). I'll get some more details on how it works later as well as a frequency color map for reference.

If no VTX device is connected the LED will revert back to the user defined color.
[image]

Betaflight Fork: branch vtx-led-strip
https://github.com/sheaivey/betaflight/tree/vtx-led-strip

Betaflight Configurator: branch vtx-led-strip
https://github.com/sheaivey/betaflight-configurator/tree/vtx-led-strip

mikeller, 2017-06-10:
@sheaivey: Good work! I can't really test it, since I don't have a SmartAudio VTX, but I think adding this makes good sense. I also think replacing 'BLINK_ON_LANDING' is a good choice, since this never was implemented properly, and is very disappointing as it is (think of it as 'BLINK_IF_THROTTLE_LOW').

Can you please open pull requests for betaflight and betaflight-configurator, so we can review and then merge these changes?

sheaivey, 2017-06-11:
Thanks @mikeller, PRs opened.

I added some screen shots to the PR to help explain how the VTX overlay works.

Betaflight: PR #3250 

Betaflight Configurator: PR 521
https://github.com/betaflight/betaflight-configurator/pull/521