betaflight/betaflight #3231: LEDs with `yaw_motor_direction=-1`

mr-git, 2017-06-07:
I switched the rotation direction of motors to opposite of BF default for `Quad X` and used the `yaw_motor_direction=-1` to adjust the yaw. Also, I have 8 LEDs set to show the stick position change (pith forward - blink top 4, pitch backwards - bottom 4 blink, roll left or right - 4 left or right LEDs are blinking). After setting `yaw_motor_direction=-1`, the LEDs are switching on opposite sides - left for roll right and right for roll left - same for pitch.

mikeller, 2017-09-18:
@mr-git: I tried to reproduce this, but it looks like this has been fixed by now. Can you please confirm?

mr-git, 2017-09-19:
Hm.. I cannot repeat it any more either. Maybe I had LEDs installed upside down or I had one of first BF 3.1.x versions. Then I workaround the issue by laying them out in different pattern, but I had removed them since then.
Installed and enabled them back now - seems that everything is working just fine now, with 3.1.7.

Will try to update to 3.2.0-RC5 on weekend, will see, if they behave as expected then.

Thanks for taking a time to look into the issue!

closed 2017-09-19 by mikeller