betaflight/betaflight #3245: Dshot1200

Trussroads, 2017-06-10:
I am trying to get in touch with someone who actually can help, I have 3 setups running DYS F4 and Dshot1200 and the only way to get pass through for Blheli32 is to flash betaflight 3.2 and then flash back because DYS F4 won't accept higher than betaflight 3.1.6 and as it is on OmnibusF4 target.. Can we add a target for DYS with pass through please or is there a way for me to activate Blheli32 pass through on this early target? It is seriously ridiculous having to do this process every time I need to alter timings etc.. HELP!!

Arakon, 2017-06-10:
Uhm.. if 3.2 already does have working passthrough, then the feature is clearly in the works. There's not going to be an old version with it implemented unless you do so yourself. Your only option aside from that is to wait for 3.2 to release.

DieHertz, 2017-06-10:
Omnibus F4 works perfectly on 3.2.