betaflight/betaflight #3246: Quad locks up and falls out of sky.

MiddleMan5, 2017-06-10:
Relevant Hardware:
+ OMNIBUS F4 V3 BARO
+ XM+
+ Taranis X9D

I've been trying to get this quad tuned and flying for the past month or so, but every time I take it out I get ~30 seconds (not consistently) of flight before the quad locks up and fallls out of the sky. It doesn't seem to lose power as it holds it's last angle (even during a flip) until it hits the ground (where I can sometimes still hear the motors running)

This is the same quad I was 100% sure was being affected by an SD card bug, but the SD card is empty and it's still behaving strangely. [This is the issue](https://github.com/betaflight/betaflight/issues/3088) that may no longer be relevant

I've been racking my head and I can't find a cause, I've checked the BB logs but I can't find anything more than "rxSignalRecieved = 0" sight before it crashes. I'm sorry I can't provide more information.

I had to add a .txt extension as github wasn't allowing me to upload zips or BFL files
--------------------------------------------------------------------------------------------------
[GoodLog.bfl.txt](https://github.com/betaflight/betaflight/files/1065876/GoodLog.bfl.txt)
[BadLog_1.BFL.txt](https://github.com/betaflight/betaflight/files/1065878/BadLog_1.BFL.txt)
[BadLog_2.bfl.txt](https://github.com/betaflight/betaflight/files/1065877/BadLog_2.bfl.txt)

jflyper, 2017-06-11:
@MiddleMan5 
OMNIBUS F4 V3 requires post-3.1.7 firmware to operate correctly on inverted serial RX. Can you test a recent nightly build?

Ref:
https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4

RipperDrone, 2017-06-11:
Have an Omnibus F4 V3 as well, does same thing on BF 3.1.7. Sometimes after punches, sometimes just after a flip or roll: Sounds like desync, motors just stop most of the time, no restart, quad just crashes into the ground. Could be false failsafe kicking in (due to RX signal failure?). Have updated to Nightly just now, will test fly next week. Also switched SD card recording off, this was unstable in some interim builds as well afaik.

Voltage supply should be ok: Battery levels were full to half full when crashed happened, I LiPo-power the FC and only run the RX voltage supply off the FC. VTX+Cam powered from LiPo.

richard-scott, 2017-06-16:
In this [issue ](https://github.com/betaflight/betaflight/issues/2430)they say:

> I have similar issue with KISS24A DShot and BetaflightF3.
I rise digitalidleOffsetPercent from 4,5 to 7 and was the same. So I rise it to 9 and now i don't have desync at all, but i think my idle throttle is little to high (than that I'm used to).

Not saying its the right answer but you never know it may help.

yujanshrestha, 2017-06-19:
My omnibus f4 board was acting very strange and I have the same issue. Random failsafes kicking in. RSSI is just fine and never get any warnings. When I pushed down on the board near the uninverted UART pads I was randomly getting SBUS (inverted) to work. Tells me something is not properly pulled up / down or something. I tried resoldering all the pads on the F4 chip with a fine tip iron and no luck. 

However, I did the XSR hack to get at the uninverted SBUS signals and threw it onto an uninverted UART. I have not had the issue occur yet but I have not done extensive testing. I will tonight and let you know how it turns out.

nathantsoi, 2017-06-19:
Do you guys have airmode on? How is your xsr powered?

RipperDrone, 2017-06-19:
Yes, airmode on. Went flying with BF 3.2 Nightly flashed on the FC - looked pretty stable this time. Maybe 3.1.7 is somewhat flawed?

yujanshrestha, 2017-06-20:
For what it's worth the sbus inverter hack worked for me. Glad to see it is fixed in 3.2.

MiddleMan5, 2017-07-01:
So let me reiterate, this has been my go-to quad, I've never updated it past 3.1.6

Fpv4tw, 2017-07-15:
My Quad locks in the air, too. And it drives me crazy.
I already changed 2 ESCs, the pdb, the FC and the OSD. 
Now my Setup is:
Carbon Blackout mini h frame
Omnibus F4 Pro V3
pdb matek
Frsky XSR 
FT952 VTX
Pz0420h cam

Log of last flight with crash:
[https://www.dropbox.com/sh/nv6e7yvzkf66bmv/AACJqr1EbWYBNZ8UqJItLlsha?dl=0](url)
In the end is the crash (obviously duh) and I immediately disarmed after I saw it already falling down to prevent any further damage as it began in about 10m height. Sometimes there is a short shake that the whole quad makes and shortly after (not measurable) it locks with motors almost dead and falls down.

Oddly enough I burned one battery prior to that in angle mode.
This flight began in angle mode, ended in manual.

Maybe as a sideinfo: The videopicture is not clean. I.e. horizontal stripes depending on the amperage used, although the OSD, CAM and VTX should be filtered. I also added capacitors on the ESCs and the OSD.

I'd be more than happy if this is solved somehow, otherwise I don't know what to do anymore.

MiddleMan5, 2017-07-16:
I should of responded with my update.

I tried upgrading to a 3.2.0 nightly and have not had the issue since. It's not exactly scientific, I haven't flown it in the exact same method since.

I've also changed:
- Practice Fields
- Disable SD card logging
- Physically removed SD card

My thought on the field is that it's possibly some kind of strange radio interference due to all of the fricking cell towers in the area.

Although it could also be related to SD card logging as I noticed odd behavior with logging enabled a while back.

Since the upgrade I've been flying it hard, maybe 50 flights and haven't had a single issue other than having to retune filters and PIDs.

*NOTE*
You will need the 3.1.2 version (or later) of betaflight-gui to interface with the new 3.2.0 firmware.

Firmware I used here:
(placeholder)
Betaflight-GUI
(placeholder) 

Give me a sec I'm on mobile

Fpv4tw, 2017-07-16:
Hi Middleman5,
I'll try your version and leave a comment about how it flew.
Gimme a week to find some time to do so ;)

Thx and have a nice week.

aguindehi, 2017-08-16:
I seem to have the same problem. Using a Omnibus Pro V3 I first run into the sdcard problem, disabled sdcard logging and then run into this problem. Updating to 3.2RC3 resulted in the insight that I always see a row of stars or a "rx lost" message in the osd just before a crash (and sometimes during flight). I also had it during a punchout, where I even noticed that I was not able to reduce throtle for a while.

Did you find a solution to the problem in the meantime?

mikeller, 2017-08-16:
@aguindehi: The 'row of stars' is the visual beeper. This means that the flight controller is 'beeping' because something is not right.

DanNixon, 2017-08-16:
@aguindehi Then there is something wrong with your RX.

The "RX LOSS" message indicates that your receiver is in failsafe mode or the FC has not received a packet from it in enough time.

Make sure whatever you are powering it from is not browning out, your antennas are in good condition, etc.

aguindehi, 2017-08-16:
Ok. Taking the hint of "rx loss" and not knowing what could be wrong with my reciver, I started replacing the receiver stuff randomly. Luckily I pulled my receiver cable first and replaced it with a brand new one freshly made with a new picoblade plug.

I've now flown the copter the whole afternoon and I had no crash. So fingers crossed it really was a simple cable problem (hurting me for 3 weeks now). *grrr*

Sorry for the noise here, I hope my problem has really disappeared! I will followup when not.

@mikeller: Thanks! Now it all makes sense. ;)

Jefflesh, 2017-09-06:
Any update on this? Are you still having issues?  I'm asking because I seem to be having an issue with most of my quads.  All have BF 3.2 (flashed to RC4 last night and still need to test).  My 5" quads are running DYS Omnibus F4, various motors and esc.  Receivers are FrSky XSR.

On my tiny whoop I have a betaFPV F3 board, was doing some testing with it tonight.  Again flashed to BF 3.2 RC4.  I ran 4 batteries through it, 2 batteries were ok, and 2 had a failsafe.  In the case of the failsafes, the RSSI never dropped below 80 (the quad whoop was just a few feet in front of me).  Telemetry drops out and the quad goes into a failsafe. 

I'm getting a little nervous to fly as I'm losing confidence in my gear or my ability to build!

aguindehi, 2017-09-06:
I had no problems anymore. Didn't cash since replacing that receiver cable. I had one time where I had the feeling I've fallen into stage 1 failsafe on a punch out for a very short period of time, but I had no message and didn't see/hear a beep, so I'm not sure I did not imagine it.

mahfax, 2017-11-28:
Having the same issue with my budget build but using a Flysky FS-i6 Tx and the X6B Rx. Ended up replacing the whole quad as well as the receiver and transmitter but still the same issue :( slowly losing hope. Any help would be appreciated. Full build list below.

RS 4in1 20A ESC
RS 2205 2300KV Motors
SP Racing F3 FC
Matek PDB
Flysky FS-i6 Transmitter
Flysky FS-X6B Reciever

Always seems to drop towards the front left motor if that's any help.
Thanks!

jflyper, 2017-11-29:
For those of you still having a problem of "falling out of the sky" problem, it usually is a hardware issue. Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

Closing.