betaflight/betaflight #3248: Detect frozen PPM values as no RX signal

thisismyrobot, 2017-06-10:
I've found some of the cheap all-in-wonder PPM Flysky RX's (eg the RX in the FLF3_EVO_BRUSHED from Banggood) don't communicate loss of RC signal to the FC. Upon loss of signal the receiver just keeps sending the last received value on all 6 channels - the ultimate fly-away mode. These RXs don't apply the TX-configured failsafe PPM values either (though they do flash a helpful blue LED when the RX is out of range?!).

I have [coded up a basic workaround in rx_pwm.c (against 3.1.7)](https://github.com/betaflight/betaflight/compare/master...thisismyrobot:ppm_freeze_3.1.7_patch#diff-5a9bf24388376542cd4115a5141145e7) that detects 3 seconds of no (significant) stick movement and forces isPPMDataBeingReceived() to return false and it works really well.

Does the Betaflight community see value in this sort of enhancement?

If so I am happy to go away and re-write it to actually fit the styles/conventions used in your codebase (my c is very rusty too), add CLI parameters to enable/configure and come back with a polished PR.

I also think this feature needs to only work when armed - so maybe instead of changing the behaviour of isPPMDataBeingReceived() I could add an isPPMDataFrozen() to rx_pwm.c and call it from in rx.c, taking into account whether we're armed or not - something like:

```c
//rx.c
...
    if (feature(FEATURE_RX_PPM)) {
        if (isPPMDataBeingReceived() /* new stuff: */ && !(isPPMDataFrozen() && areWeArmed())) {
            rxSignalReceivedNotDataDriven = true;
            rxIsInFailsafeModeNotDataDriven = false;
            needRxSignalBefore = currentTimeUs + needRxSignalMaxDelayUs;
            resetPPMDataReceivedState();
        }
    }
...
```

Anyway, I'm very keen hear your feedback, worst-case I just keep patching the BF releases for myself :)

jflyper, 2017-06-11:
I would say FrSky should fix the board, recall and replace every unit they've sold.

thisismyrobot, 2017-06-11:
@jflyper I'd agree, except it's as-advertised if you look close enough at the "features" - which I didn't at the time.

You make a good point though, this enhancement only makes sense if PPM continuing to send the same value for a period of time is a state that is considered "bad", regardless of cause.

thisismyrobot, 2017-09-15:
Closing due lack community interest and not wanting to leave stale rubbish amongst all the proper bugs and feature requests in here :)

closed 2017-09-15 by thisismyrobot

mikeller, 2017-09-15:
Agreed with @jflyper in that we should not take on responsibility for covering up serious safety issues.

@thisismyrobot: For most RX out there, the 'no signal' behaviour can be configured, and the most common procedure for configuring the RX to stop outputting a signal in case of RX signal loss seems to be to press the 'bind' button for a couple of seconds, while the RX is powered up but not receiving any signal. Have you tried this with your RX?

thisismyrobot, 2017-09-15:
@mikeller After trying to configure the RX via my TX, and via the bind method, I looked closer at the specs and the RX (in small text) doesn't support Failsafe at all. They don't support any form of RX signal loss notification (either by stopping sending PPM pulses or sending something out of range).

Upon loss of signal all it does is keep sending the last PPM signal it got. Forever :(

That's why I wrote the code I did to work around a flaw in what is simple a cheap and nasty RX. My thought was that with the proliferation of these in all the sub-$100, sub-100mm brushed micros, maybe this would be a valuable addition to BF. I also figured I wouldn't be the only sucker to buy one without reading the fine print.

That said, now I've asked the question and gotten your feedback, I agree with you and @jflyper - it's not up to BF to work around shoddy unsafe hardware so I've closed this issue.

Noctaro, 2017-09-15:
I wouldnt call it responisbility. Creating a workaround for this kind of setup would just help users that were trapped with some crappy rx, more sounds like a saftey feature to me. I also have one of these boards. You can not force them into fs mode. Lucky its only a brushed copter, but its attemps to flyaway are real. This feature would have helped. However i know you guys are doing alot of awesome things in this code and i also understand that this feature is very low priority. But wouldnt it be a nice to have at some point of the history? :)

@mikeller  i also tested your suggestion on my board, but its just crap and cant handle any other fs mode then crazy signal repeating

mikeller, 2017-09-15:
@Noctaro: @mikeller i also tested your suggestion on my board, but its just crap and cant handle any other fs mode than crazy signal repeating. How cheeky....

Sorry to say, but the only way forward from this is to throw away this RX, and buy one that's not brokenin a way that makes it dangerous to use. If this was a mains power cable, and you found it was defective and one of the wires was exposed somewhere along the cable, would you keep it and 'fix' it by wrapping a bit of string around the exposed bit?

Noctaro, 2017-09-16:
@mikeller 
Yes, i agree with you, such a rx in a multirotor setup is just a nogo. I received a replacement with a new version of the rx, when i mentioned my problem at a conversation with the seller. This new one also supports sbus/ibus. So the producer did listen and solve the problem. Maybe some of you who are trapped with those components can open a claim. This might also prevent future hardware fails like this. 

I dont know. Even back in time when i flew airplanes, i did not like the repeat function. In early FPV days its also was no fun when your plane just went into a far vacation flight. Was so happy when i bought my first 2.4 frsky setup, where i could disable this feature. Of course i can imagine it makes sense in some scenarios.

mikeller, 2017-09-16:
@Noctaro:

> Or has it been a hardware limitation?

I guess so, probably a remnant from the days when RC radios were still entirely analogue, and the signal just 'went away' and left the servos wherever they were at the moment.