betaflight/betaflight #3259: [Discussion] Pin (resource) locking

jflyper, 2017-06-12:
For discussion.

This is a continuation of the discussion about pin locking feature/mechanism that has been started in the context of reconfigurable I2C (#3229).

jflyper, 2017-06-12:
https://github.com/betaflight/betaflight/pull/3229#issuecomment-307287943
> The last commit has a change to give an idea about how we handle "Non reconfigurable pins" (at least as long as user has a chance to manipulate pins directly through cli commands).

This is the change.
```
// Non-reconfigurable pins
 +#define LOCKED_IO_PORTA         (BIT(9)|BIT(10))  // PA9 and PA10 for UART1
 +#define LOCKED_IO_PORTB         (BIT(10)|BIT(11)) // PB10 and PB11 for I2C2
 +#define LOCKED_IO_PORTC         0
```

>
>Background of this idea is to prevent critical functions from accidentally remapped to something else and wedging boards. The example is NAZE, in which I2C2 (PB10, PB11) can't be touched because it has gyro on it, and UART1 which is used for MSP. (There may be others, these are just examples.)

mikeller, 2017-06-12:
In a way, after the discussion, and the realisation that it will probably be hard to get a 'watertight' system in place (i.e. resources being initialised first 'stealing' pins that are required by critical resources that are initialised later on), I have started to wonder if this is really required. There are other config settings that can wedge a board, but fortunately, the STM32 boot loader mode makes it pretty much impossible to wedge a board in a way that is not recoverable.

martinbudden, 2017-06-12:
I tend to agree with @mikeller on this.

ledvinap, 2017-06-12:
I can imagine implementing locking as useful warning system. 
But personally I'm OK with burning my fingers when I do something stupid ... (not cutting them by propeller)

ledvinap, 2017-06-12:
It would be better to handle this completely in Configurator .. firmware may just check collisions and bail out on first error.

mikeller, 2017-06-12:
Re doing this in configurator, the problem with that is that we'd need to make the configurator aware of the hardware specs of all 100+ targets. Considering how hard it already is to get collaboration from manufacturers for the support of their targets, I don't see this as a realistic goal.

ledvinap, 2017-06-12:
@mikeller : I can imagine holding all necessary info in one place, with automatic target.h generation.. It may make management easier ...

mikeller, 2017-06-12:
That is an option. But even that would make it harder to work with manufacturers that already now struggle to put useable target definitions together...

martinbudden, 2017-06-12:
The trouble with this is that it is yet more target specific stuff to maintain. It's extra work for us and of marginal use to users.

ledvinap, 2017-06-12:
IMO parameter groups have to go first anyway, maybe it will be easier then ...

It may make sense to use 'LOCKED_IO_PORTx' for now (with all disadvantages) and only on well-supported targets. Implementation should be quite easy and it will be localized to CLI, making it easy to replace later.

jflyper, 2017-06-12:
Well, then, let's leave the pin lock feature alone for a while. We can go for the `LOCKED_IO_PORTx` like @ledvinap said above if we start hearing screams of curse.

ledvinap, 2017-06-12:
@jflypper: Yet another quite simple possibility is to define list of locked pins in target.h. It will handle preprocessor-defined pins a bit better ..
```
target.h:
#define LOCKED_PINS IO_TAG(I2C1_SDA), IO_TAG(I2C1_SCL), SPI1_PINS

somewhere.h/c
#define SPI1_PINS IO_TAG(SPI1_MOSI), IO_TAG(SPI1_MISO)

cli:
ioTag_t lockedPins[] = {LOCKED_PINS};
```

jflyper, 2017-06-12:
@ledvinap That's another way of doing it. A possible empty initializer may be a problem, but we can get away somehow;)

closed 2017-07-09 by jflyper