betaflight/betaflight #3261: Transponder not supported on BrainRE1

Zethyl, 2017-06-12:
The betaflight configurator indicates that the transponder is not supported on the board (BrainRE1). This is the only feature keeping me from going on Betaflight with this board, everything else seems to work.

mikeller, 2017-06-12:
The Betaflight firmware for the BrainRE1 has its own repository at https://github.com/BrainFPV/betaflight/issues , so you're going to have to bring this up there. But in principle, Betaflight has support for transponders, so unless it's a hardware issue, there is nothing preventing Betaflight on the BrainRE1 from supporting transponders.

jflyper, 2017-06-12:
One thing to note is that transponder support for F4 is relatively new (introduced Mar. 1, 2017).

hydra, 2017-07-02:
I'll just chime in and say the F4 transponder code works fine on the SPRacingF4EVO and SPRacingF4NEO with iLap (the fastest transponder protocol that requires the best hardware).