betaflight/betaflight #3263: Nero F7 bf3.2 arming with led strip enabled locks fc

Cheredanine, 2017-06-12:
As per title, couple of issues already logged about LEDs misbehaving with F7s but not this behaviour, 
Running led strip from motor pad 5 as per target wiki

Even just basic colour on LEDs.

LEDs work fine, but as soon as armed fc freezes.

Tried shifting motor 5 resource to none, no difference

jflyper, 2017-06-13:
FYI, the symptom does not develop with OMNIBUSF7. Nero specific?
(Still flickers t

jflyper, 2017-06-18:
Tested with Nero. Works fine.

Cheredanine, 2017-06-18:
Hmm,
Pretty confident of wiring, LEDs work fine on various settings before arming, even arm state indicator works, fc just locks after that. Will retest with latest master
(Also using m5, do have bb and sd enabled, but turning led off and on changes the behaviour)

Cheredanine, 2017-06-18:
@jflyper 
Ok here is the only delta between you and me in setup (ignoring the possibility of error)

Jason based the Nero on his bluejay f4, the f4 used uart6 for sbus (uart3 did not have an external inverter), now whilst the inverter is built in on f7, the hysical layout of uart 3 (pin order) is still configured to match a microminim and uart6 is suggested for serial Rx or ppm
And that is what I have used (and continue to have the issue)

jflyper, 2017-06-18:
What if you turn off BB & SD?

Cheredanine, 2017-06-18:
Just tested, same,
Only happens with led strip feature enabled
Appears to be independent of led settings (Larson, colour etc)
Fc locks on arming, if configurator is connected freezes on current screen, at field, motors don't spin up
Arm state indicator on led (if set) will change to indicate arm state (although very belated) but will not change back on disarm - further suggesting fc is locked up

mikeller, 2017-06-18:
Could this be a problem of the LED strip drawing too much power on your 5V rail? Can you power the LED strip externally for a test, and see if that changes anything?

Cheredanine, 2017-06-18:
Seems unlikely, happens when bf is connected and laptop providing power
5v/600mA pololu powering pc, xm+ and 3leda, identical setup with bluejay is fine, 
All working and fictional when dissarmed

mikeller, 2017-06-18:
Even unlikely causes are worth being eliminated, if there is an easy way to do so. ;-)

Cheredanine, 2017-06-18:
Well since it would entail stripping the quad down, desoldering the led strip, finding a new source, resoldering, testing, then reversing the process, it is not that easy, however, I am game, but timeline is likely to be Thursday evening at the earliest

Cheredanine, 2017-06-18:
Edit, had soldering iron out anyway
Stripped quad.
Disconnected 5v line from LEDs so LEDs disabled, behaviour persists, if I enable led strip fc locks on arming
Also tried os42, same behaviour

Cheredanine, 2017-06-30:
Edit.
So reflashed and retested and gone away, go figure, putting down to liveware issue
Closed

closed 2017-06-30 by Cheredanine