betaflight/betaflight #3268: BFF3 3.2 #1336 OSD elements showing when disabled

BludzUK, 2017-06-12:
Hi,
I'm running a BFF3 on 3.2 #1336 and the OSD has elements that have been disabled in the configurator. It also has duplicate elements.
I have all the gps data off and the horizon lines, but they are still there. Fly Time is repeated in a different position. Very strange bug.
Happy to help if I can. Here's a cli dump of just the osd named attributes.
set osd_units = METRIC
set osd_rssi_alarm = 20
set osd_cap_alarm = 1000
set osd_time_alarm = 2
set osd_alt_alarm = 100
set osd_vbat_pos = 2467
set osd_rssi_pos = 2084
set osd_flytimer_pos = 2485
set osd_ontimer_pos = 23
set osd_flymode_pos = 26
set osd_throttle_pos = 2103
set osd_vtx_channel_pos = 25
set osd_crosshairs = 200
set osd_horizon_pos = 200
set osd_current_pos = 2455
set osd_mah_drawn_pos = 2435
set osd_craft_name_pos = 18
set osd_gps_speed_pos = 26
set osd_gps_lon_pos = 2130
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 25
set osd_home_dir_pos = 2350
set osd_altitude_pos = 23
set osd_pid_roll_pos = 15
set osd_pid_pitch_pos = 15
set osd_pid_yaw_pos = 15
set osd_debug_pos = 2049
set osd_power_pos = 26
set osd_pidrate_profile_pos = 25
set osd_battery_warning_pos = 2474
set osd_avg_cell_voltage_pos = 2124
set osd_pit_ang_pos = 2305
set osd_rol_ang_pos = 2337
set osd_battery_usage_pos = 2440
set osd_arm_time_pos = 2113
set osd_disarmed_pos = 2186
set osd_stat_max_spd = ON
set osd_stat_min_batt = ON
set osd_stat_min_rssi = ON
set osd_stat_max_curr = ON
set osd_stat_used_mah = ON
set osd_stat_max_alt = OFF
set osd_stat_bbox = ON
set osd_stat_endbatt = OFF
set osd_stat_flytime = OFF
set osd_stat_armtime = ON

McGiverGim, 2017-06-12:
The OSD is attached to the configurator. You must use exactly the configurator with the same elements than the firmware.

BludzUK, 2017-06-12:
I'm using the configurator to disable elements I don't want, they don't show in the configurator preview but they still show in my video feed. I don't quite understand what you are saying, can we not configure osd in the configurator? Only in the firmware?

DanNixon, 2017-06-12:
You must be using the latest (i.e. `master` from the repository here: https://github.com/betaflight/betaflight-configurator) when using nightly builds of the firmware.

McGiverGim, 2017-06-12:
If you use a beta version of bf, for example the actual master, you must use the beta version of configurator, the actual master too.

nathantsoi, 2017-06-12:
Or you can modify visible elements in the OSD via stick commands

DanNixon, 2017-06-12:
@nathantsoi Well, I may not have been keeping up with maintaining that menu, my bad.

(I do have a branch that fixes it)

DanNixon, 2017-06-12:
Also the two fly timers are different (we just don't have separate icons for them, that will be fixed when the OSD fonts are overhauled, eventually).

One counts all time the craft is armed (fly time), the other counts time since last arming (arm time).

BludzUK, 2017-06-12:
Thanks for all the help guys. I will reflash and setup using the correct configurator. Haven't flown a nightly since dshot600 all those months ago, lol.

mikeller, 2017-06-12:
@DanNixon: Speaking of arm time / fly time, do we need both of them, or would it make more sense to have only one timer, and an on/off setting for `showTotalArmedTime` to toggle how it works?

DanNixon, 2017-06-13:
@mikeller Maybe not.

In fact it may be better to have "generic" timer elements (something like `TIMER1` and `TIMER2`) and have settings to define what they are for, this could include on time, total armed time, last armed time, on time when disarmed and armed time when armed, etc.

BludzUK, 2017-06-13:
Solved. Thanks again for the advise

closed 2017-06-13 by BludzUK

mikeller, 2017-06-13:
@DanNixon: That's what I was suggesting: Have a generic timer element, and let users select what it should display, instead of having one element for every single timer.