betaflight/betaflight #3274: 0v switch

mattyra, 2017-06-13:
Hello world......

Is it possible to have a 0v out switch ?? 

I have a VTX with a push button that just needs the gate colosing... 1 second change channel 3 seconds band 5 seconds power so I thought is it possible to get this direct from the FC ?? 

I had originally though about using a MOSFET switch and maybe changing using the led out but thought before I run away with it I should check with the Gurus......

Just to be clear I'm looking to take away the push button solder on 2 cables that can the act as the switch that are controlled via the FC

Yes I know it will be hard to get the timming right but maybe it's a cool hack... I'm sure some one will have already thought of this, had a quick search online but was unable to uncover any thing that's moved with the times......

Most my quads have pods so changing is a real pain at times.... Could change to all new VTX with smart audio would be cool to this hack.......

jflyper, 2017-06-13:
@mattyra There are two options.

(1) It is hard to create a switch without some external components (such as a MOSFET which you have mentioned). A software module can be written to drive this circuit.

(2) Use PWM controlled switch.
http://www.hansenhobbies.com/products/onboardacc/ers/pers/

mikeller, 2018-06-03:
@mattyra: I think this was implemented in #5108. Can this be closed?

closed 2018-06-10 by mikeller