betaflight/betaflight #3288: Serialpassthrough on CP210x devices appear to only use the baud rate of the MSP

brycedjohnson, 2017-06-14:
I was testing programming the xsr over sport using the serialpassthrough.  On an omnibus board with VCP it worked with no issue.  It communicated with the frsky tool at 57600 as desired.

With the RMDO board it wasn't communicating.  It turns out that when I set the MSP to 57600 and used serialpassthrough it worked correctly.  Is that a limitation of the chip or a bug?

ledvinap, 2017-06-14:
Just guessing, but it may be problem with flow control. 
Pass-through should use configured baud rate for port

brycedjohnson, 2017-06-14:
@ledvinap With the MSP configured to 115200 with serialpassthrough for the sport port (57600).  I looked at the data in putty, it didn't look right at 57600 but it was correct at 115200.

When I set the msp to 57600 the data looked right at 57600 and programmed correctly.

I am using the master branch as of a couple days ago for the build. Not 3.1.7, maybe there was a change?

mikeller, 2017-06-14:
One of the things to keep in mind is that for boards using VCP, the MSP baud rate is ignored.

From your experience, it looks like flow control works on VCP though, whereas it does not seem to be working on CP210x. Not sure if the pins for hardware flow control are even connected between the CP210x and the MCU on any of the boards using a CP210x.

mikeller, 2017-06-14:
> With the MSP configured to 115200 with serialpassthrough for the sport port (57600). I looked at the data in putty, it didn't look right at 57600 but it was correct at 115200.

That was between PC and flight controller, or flight controller and RX?

brycedjohnson, 2017-06-14:
I haven't deal with hardware flow control so I'm not quite sure how that
works.

The data was between the FC and the PC.  The initial blob that I saw on the
CLI looked right when I enabled passthrough, but when I disconnected​ and
looked it out though putty it didn't look right anymore and the frsky tool
didn't work.

mikeller, 2017-06-15:
@brycedjohnson:

> I haven't deal with hardware flow control so I'm not quite sure how that
works.

For hardware flow control, 2 extra lines (one in each direction) are used for the receiving side to signal to the sender when it is ready to receive data. If these lines are not connected between CP2101 and flight controller, and the UART speed there is 115200, and the UART speed for the RX is 57600, then the PC will be sending data at twice the speed that the FC can forward the data to the RX, without a way to tell the PC to wait until it has cleared the backlog. Once the buffer on the flight controller is full, this will cause an overflow.

closed 2018-04-21 by DieHertz