betaflight/betaflight #3290: Wiki controls help page missing

valerioangelici, 2017-06-15:
The page that could be found here
https://github.com/martinbudden/betaflight/blob/master/docs/Controls.md
is not in the wiki, could it be possible to add it possibly with a graphic representation of stick positions for both mode 2 and for us poor mode 1 users?

mikeller, 2017-06-15:
There is a link to this documentation right in the very first paragraph of the Betaflight wiki.

closed 2017-06-15 by mikeller

valerioangelici, 2017-06-15:
a bit hard to find, I think them should be handy to find when needed, and a mode 1 graphic representation should be handy too even if there are not so many mode 1 users.

mikeller, 2017-06-15:
The image is here. And I think a mode 1 version would be welcome, a quick search on the internet did not turn up one. 

https://github.com/martinbudden/betaflight/blob/master/docs/assets/images/StickPositions.png

Once you've done the mode 1 version, feel free to add a 'Stick Commands' entry to the 'Setup and More' category in the wiki with the 2 cheat sheets - it is user editable.

valerioangelici, 2017-06-15:
> StickPositions.png

ok thanks, I was asking if there was a photoshop source file to make the
edit a bit simpler instead of making cut and paste over the png image, in
the assets/images folder there are two files, a png and an svg so I was
wondering if there was somewhere the original file from which these have
been exported.