betaflight/betaflight #3291: Dshot 600 on Colibri race v2 w/ TBS25A blue caps - motors twitching hard

ITPrincis, 2017-06-15:
Hi all,
don't know if I posted this in right place...
I have come across problem. I have Colibri Race v2 and 4x TBS 25A blue cap (dshot600) capable escs. When I use oneshot42 everything is fine, when dshot600 above approx 1150us motors start twitching hard. 
I haven't tried escs with different FC.
Could this be target (BF code) issue? As TBS says their 25A escs work with dshot600 if they have blue cap on.  ESCs on BLHeli-S 16.65 default settings except throttle set to 1000-2000.

I have attached video for reference: 
One video with oneshot, everything runs ok, no twitches. https://drive.google.com/file/d/0B6BIaGgUttA6NFRqYUxGVkZTVVE/view?usp=sharing 
And here with dshot 600, when I apply throttle around 1200us motors start twitching...  https://drive.google.com/open?id=0B6BIaGgUttA6SndiX1BXaVh4Rmc

BayaroX, 2017-06-21:
Hi!

My TBS FPVision Elite Bundle with Colibri Race v2 FC are working fine with DShot600. But i have big problems/twitches with DShot1200 on 2 Quads.
I use Airbot Wraith32 BLheli32 ESC.

But maybe it is because of the bugs with BLheli32 passthrough/communication - where is only working version 32.0.0.1 - with version 32.0.0.4 i can`t connect to my ESC.

Kind regards,
Chris

amivit, 2017-06-25:
I'm flying a Eachine Wizard X220: Blheli_S 16.65, & F3 6DOF BetaFlight 3.1.7. Tried using Dshot600 with extreme twitching as well. Unsure if there are caps I need to remove, but I went back to oneshot125.