betaflight/betaflight #3293: Lowest cost microcontroller possible

whmountains, 2017-06-15:
I'm trying to build a very low-cost, all-in-one quad.  Similar to the CrazyFlie, but more like in the $35-$50 price range.

Originally I wanted to implement it on an ESP32, but according to #2970 betaflight requires microcontrollers from STMicroelectronics.  Further research leads me to believe that it needs to be in the STM32F3 or STM32F4 series.  Is that correct?  Are there any other requirements?

I'm looking at [this MCU](https://octopart.com/stm32f410cbu6-stmicroelectronics-67124559) as a possible option.  Is it likely to work?  If not, what *would* be the lowest cost board available?

Thanks a lot!

mikeller, 2017-06-15:
Yes, Betaflight is currently only supporting STM32 MCUs (and SITL for simulation).

A quick look at the STM32F410 data sheet reveals the following:

> Up to 128 Kbytes of Flash memory

While it is possible to build Betaflight for targets with 128K (like the F1 based legacy boards), it requires for all non-essential features to be removed in order to make it fit. This is certainly not advisable for any new designs.

If you are looking for a cost effective MCU, STM32F303 is probably a good choice.

jflyper, 2017-06-15:
+1 for F3. Avoid F4 at all costs.

whmountains, 2017-06-15:
Thanks for the advice!

closed 2017-06-15 by whmountains