betaflight/betaflight #3296: OpenPilot CC3D Revolution + Buzzer on PIN6

mixMugz, 2017-06-15:
After update from 3.0.0 (special edition with pin6 buzzer support) i have lost control with my buzzer.
My resource:
```
resource BEEPER 1 A00
resource MOTOR 5 NONE
resource MOTOR 6 NONE
beeper_inversion = OFF
beeper_od = ON
```

jflyper, 2017-06-15:
What's the version you are having problem with?

mixMugz, 2017-06-15:
Oh. 3.1.7

jflyper, 2017-06-15:
What about the "special edition"?

mixMugz, 2017-06-15:
http://andwho.sytes.net:8080/job/AndWho_Betaflight_BUZZERM6/

jflyper, 2017-06-15:
Try this in CLI
```
resource beeper b4
```

mixMugz, 2017-06-15:
Not work. Always beep.

jflyper, 2017-06-15:
What if you make it beep (e.g., by defining BEEPER in mode tab)? Does it stop beeping?

mixMugz, 2017-06-15:
Nope.

mixMugz, 2017-06-15:
> It may or may not work (LED1 is still on PB4).

LED on A01

mixMugz, 2017-06-15:
This how my buzzer is connected to MOTOR6 out
http://i.imgur.com/zhJXLBv.jpg?h=X0bszi16Tfqxn2TjjanG6w

mikeller, 2017-06-15:
Try `resource beeper a0`.

mixMugz, 2017-06-15:
`Try resource beeper a0.`

Already... Read first post>

mikeller, 2017-06-15:
Ha, not enough coffee yet. Have you tried variations of `beeper_inversion` and `beeper_od`?

mixMugz, 2017-06-15:
yep.

mikeller, 2017-06-15:
And CLI `resource list` shows the BEEPER on A0?

mixMugz, 2017-06-16:
> And CLI resource list shows the BEEPER on A0?

YES!

mixMugz, 2017-06-16:
This is my `resource list`

```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: BEEPER 
A01: LED_STRIP 
A02: MOTOR 4
A03: MOTOR 3
A04: MPU_CS 
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE 
A09: FREE 
A10: SERIAL_RX 1
A11: USB 
A12: USB 
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR 1
B01: MOTOR 2
B02: FREE 
B03: FLASH_CS 
B04: LED 2
B05: LED 1
B06: FREE 
B07: FREE 
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE 
B12: FREE 
B13: FREE 
B14: FREE 
B15: FREE 
C00: INVERTER 1
C01: FREE 
C02: ADC_BATT 
C03: FREE 
C04: MPU_EXTI 
C05: FREE 
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE 
C09: FREE 
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE 
C14: FREE 
C15: FREE 
D02: FREE 

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: LED_STRIP
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mixMugz, 2017-06-17:
Closed:

Work perfect with this settings:

```
resource BEEPER 1 A00
resource MOTOR 5 NONE
resource MOTOR 6 NONE
beeper_inversion = **ON**
beeper_od = **OFF**
```

closed 2017-06-17 by mixMugz