betaflight/betaflight #3298: Hard yaw causing throttle punch

dronebrulee, 2017-06-16:
Fine tuning the roll/pitch/yaw rate curves for my 6" LOS acro quad to get a natural feeling, quick 360deg. flip/spin.  Happy with the roll/pitch curves, but experiencing weird behavior with yaw: whenever I apply hard yaw stick deflection to get a quick spin, the throttle also inexplicably punches on it's own.  I am setup for mode 2 so the yaw and throttle are together on the left stick, but this is definitely _not_ being cause by sloppiness on the stick where I am inadvertently applying throttle at the same time as the yaw - I thought that might be the case at first too so did some focused experimenting - it really is punching directly as a result of the hard yaw alone.  Any insight whether this might be a known scenario with Betaflight?  Or maybe something flaky with the gyro hardware feeding into Betaflight that would cause it to jump the throttle?  I suppose it could even be an issue with Tx firmware, though that seems less likely?  Thoughts?

Betaflight 3.0.1
Warthox MiniCP120 Frame
Airbot Omnibus F3 FC
Racerstar RS20Ax4 20A 4-in-1 Blheli_S ESC
HyperLite V2 2205-2300 Motors
Kingkong 6040 Props
XK X7 8Ch Tx running Futaba S-FHSS protocol

mikeller, 2017-06-16:
Hi @dronebrulee.

GitHub is for discussion and questions around the development of Betaflight. For configuration related questions, have a look at RCGroups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

Also, updating to a more current version of Betaflight might help solve your problems.

closed 2017-06-16 by mikeller

dronebrulee, 2017-06-16:
Hi @mikeller - I appreciate the link, but this really isn't a request for help configuring Betaflight.  I'm experiencing significant anomalous behavior so am asking whether this might be a known problem with the firmware alone or with the firmware in conjunction with the FC sensors.  I will update to a more current version at some point (not always a simple transition) but would like to understand the nature of the problem regardless, particularly if the firmware is involved.  Would there be a problem leaving the question open for a bit?

mikeller, 2017-06-16:
You are asking for a known issue for a firmware release that is 9 months old and superseded by 8 other releases. This is hardly related to _development_.

dronebrulee, 2017-06-16:
Hardly?  Really?  So you have full knowledge of the issue, know it to have existed in 3.0.1 and then fixed in a subsequent release?  I've been a software engineer for 30+ years mike and this is a more-than-legitimate query around potentially anomalous software behavior.  I really wish you would let others with more insight consider the question.

mikeller, 2017-06-16:
Ok. So you are a software engineer. Does 'please verify this issue against the latest release of the software' sound familiar?

dronebrulee, 2017-06-16:
Not really, no.  What would be familiar is "please upgrade to the current release where this known issue has been fixed".  You're rejecting a report with no insight into whether it is or isn't an issue in the original software release or the latest one.  I really like the deep curiosity and innovation that drives Betaflight development, but I'd be sorry to learn that your response here really is the baseline attitude towards software quality/stability in Betaflight?

mikeller, 2017-06-16:
I did ask you to upgrade to the latest version in my initial response.
I did say this 'might help' solve the issue.
What is your expectation here? That an open source project that has the goal to deliver a bleeding edge solution does active bug fixing on a release that has been superseded by 8 others? The reality is, there have been hundreds of fixes and improvements in the time since 3.0.1 was released, a considerable number of them in the PID controller. This makes discussing a bug in 3.0.1 without checking if it has already been fixed in the latest release look a bit moot.

dronebrulee, 2017-06-16:
Right, it's the '_might_ help' assertion that's the problem - i.e. you asked me to upgrade, but as an uninformed hail mary that perhaps this is no longer an issue, not because you know the current release is "where this known issue has been fixed".  I'm not sure what you mean by "active bug fixing on a (prior) release" - it's not like I found this in a 1.x or 2.x release and, regardless, you are not maintaining independent 3.x point releases and backporting bug fixes to prior releases and I have zero expectation you would.  My expectation here is simply that my question stand for consideration by others with deep insight who 1) might be curious to hear about this oddness or 2) can confirm this as a known issue that has been fixed.  You don't seem to be curious in that way and you're not confirming the existance or non-existance of this issue, you are simply rejecting the question as unwelcome.  But perhaps by saying "bleeding edge solution" you're indicating that questions of software quality/stability are legitimate only for the current release.  Personally I wouldn't see that as good practice - I would accept any questions related to 3.x - but it's certainly not an unreasonable policy if in fact it _is_ the policy.  I'd just like to know if that's true and if so, I would be sorry to learn that.  So... is that the idea here, generally?

mikeller, 2017-06-17:
@dronebrulee:

> Right, it's the 'might help' assertion that's the problem - i.e. you asked me to upgrade, but as an uninformed hail mary that perhaps this is no longer an issue.

Yes I did. To me, from the point of view of fixing the problem that you are experiencing, and of identifying if this might affect other users, this is the logical next step.

Maybe I was a bit too quick in closing this issue. But still, if you are not willing to update to latest, there is not much that can be done to address your problems, so there is no point in reopening it.

If you are interested in first trying trying to figure out if and where your problem might have been fixed, feel free to do so:

https://github.com/betaflight/betaflight/compare/betaflight:v3.0.1...master
(3352 commits is a lot of ground to cover though)

> You don't seem to be curious in that way and you're not confirming the existance or non-existance of this issue, you are simply rejecting the question as unwelcome.

Yes I am curious, but only *if* the issue has not already been fixed. But you seem to be unwilling to work towards confirming this.

> But perhaps by saying "bleeding edge solution" you're indicating that questions of software quality/stability are legitimate only for the current release.

Yes, that is correct. As you already stated, Betaflight does not do maintenance on old releases, and bugfixes are generally only done on the latest release.

martinbudden, 2017-06-17:
@dronebrulee , betaflight has a small number of active developers and we simply do not have time to check bugs reported on older releases.

It would be extremely helpful if you could check if the bug still exists in the latest release - if it does then it can be investigated.

dronebrulee, 2017-06-17:
hi @martinbudden - determined it's unrelated to Betaflight: throttle punch due to high-rate yaw acceleration occurs naturally it seems and is exacerbated when motor deceleration is comparatively weak relative to acceleration - e.g. when running larger props on smaller motors, in my case running 6" on 2205/2300kv.  that said, i think the firmware could potentially defend against yaw-induced throttle punch by governing the yaw rate if it is sensed to be lagging significantly behind the expected rate as it accelerates towards the intended rate.  but i doubt that's particularly relevant as a high-performance mini-quad feature goal. thanks though for your follow-up - cheers.

brandonsD, 2018-08-10:
I’m experiencing the same issue with my quad. Yaw to the right climbs in throttle and yaw to the left declines even with throttle stick up or in a medium position it will do that. I’m running cc3d Revo on qav250 with a r9d receiver and a at9 transmitter. I made sure everything is up to date and reset everything on the program side of things but not my transmitter. Please some one help.

brandonsD, 2018-08-10:
Also when I just use the throttle only to lift off it goes up a little and then comes down like it lost power or something. But if I use the yaw to the right it gains throttle but turns the quad to the right along with it.

dronebrulee, 2018-08-11:
dronebrulee:
> throttle punch due to high-rate yaw acceleration occurs naturally

brandonsD:
> Yaw to the right climbs in throttle and yaw to the left declines

i see no difference re: vertical throttle punch *up* on hard yaw, regardless of yawing right or left - can't quite imagine how you would see vertical *down*.  that implies the quad is *losing* overall lift when the two props involved in the left yaw accelerate relative to the other two accelerating props involved when yawing right.  the acceleration in both cases is *increasing* lift from the two props involved.  i'd expect there would need to be (overly) hard braking on the two non-accelerating props to *lose* overall lift in your left yaw scenario.  if you are running a newer version of BF (i see this on 3.0.1), perhaps the yaw control logic has changed to increase the relative braking, though i can't see how that would effect the left yaw prop control differently than the right yaw prop control?  unlike the original conclusion `throttle punch due to high-rate yaw acceleration (right *or* left) occurs naturally`, what you are seeing really *might* seem to involve some BF logic and should perhaps be looked into?

brandonsD, 2018-08-12:
I’m using LibrePilot program up to date and firm up to date. It like there is some sort of breaking going on when I yaw to the left. I have reset all settings and re-flashed everything. All is default as when I had flown before with out this issue I’m haveing now. I am using a qav 250 with emax mt2204-2300kv motors. Simonk 12a esc’s on cc3d f4 Revo board. I’m using my at9 linked to my r9d receiver via sbus and running pwm protocol on esc’s. All was good till I crashed into a tall trees branch with leaves and fell to the ground. I picked it up and took off the Lipo. Then took it to a table outside and examed it and was a pass to the look of things. I plugged back in the battery and smoke came from oe esc only. I replaced that esc and motor with new and reset everything as if I was starting g new again just to be safe. And it started doing the decent left yaw even with throttle up. And incline on yawing right only with throttle given command. I can keep a Hoover if I control throttle to a medium and roll stick left to right to keep near me. But I’m spinning during this flight to see what’s going on with yaw left and right and the power added and taken when turning. Any help will do. Thanks

fujin, 2018-08-12:
@brandonsD Please, stop. This GitHub repository is for the development of Betaflight, not support for LibrePilot - they are not even remotely related. It is also not for user support with configurations.

You are looking for this:
- https://www.librepilot.org/site/index.html
- https://bitbucket.org/librepilot/librepilot
- https://forum.librepilot.org/
- https://librepilot.atlassian.net/projects/LP/issues

I am not sure it is very actively maintained. You might have better luck with Betaflight on your flight control, but again, this isn't the place to ask for support. You will need to make use of RCGroups, the Facebook Betaflight group, or the Betaflight group Slack instance `#support` channel.

Locking.