betaflight/betaflight #3301: RSSI with XSR receiver

msmith5662, 2017-06-17:
Hello,

I have scoured YouTube and the forums, even here, and have not found an answer to my specific issue.  In fact I have found conflicting information.  I have a XSR receiver and a TBS Unify Pro HV attached to my BF3 FC.   All the functionality of the Smart Audio is operational.  I have RSSI on my Taranis.  What is the proper way to get RSSI up and running on my OSD so I can see it in my goggles.  I do not want to retransmit the RSSI from the Taranis.  Right now the display just flashes on the alarm setting.  Thank you in advance.

mikeller, 2017-06-17:
Hi @msmith5662.

GitHub is for discussion and questions around the development of Betaflight. For configuration related questions, have a look at RCGroups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

Also, see:
https://www.reddit.com/r/Multicopter/comments/4r99te/question_does_the_xsr_have_an_rssi_pad/

Connect the RSSI pad mentioned in the article to the RSSI input on your flight controller.

msmith5662, 2017-06-17:
Hi, 
The article you linked is from 2015, older than my FC, plus it refers to CLeanflight.  From my understanding you do not need to do the wiring with the BF3 FC.

closed 2017-06-17 by mikeller

DieHertz, 2017-06-17:
You do need to do the wiring from RSSI pad to RSSI input on the FC, nothing has changed, how could it?

jflyper, 2017-06-17:
@msmith5662 I'll follow you up in the RCG thread mentioned above.