betaflight/betaflight #3305: Taranis - "Sensor Lost"

carloscampos775, 2017-06-17:
Using OpenTX 2.2 the Taranis will sound intermittent warnings about a "Sensor Lost". After going through the sensors, I found the issue is with the VFAS sensor. The VFAS sensor will stop updating at random intervals. I have also discovered this issue is only happening on FCs that use a current sensing pin like the "SP Racing EVO - Betaflight" and "Revolt F4 - Raceflight". This issue is not happening on boards like the Betaflight F3 or Kiss FC that take direct battery voltage.

https://github.com/opentx/opentx/issues/5035#issuecomment-309212678

yujanshrestha, 2017-06-19:
Are you using an XSR. Try updating the firmware and checking out [this forum](https://www.rcgroups.com/forums/showthread.php?2793214-Modify-Betaflight-PID-VTX-Settings-w-TARANIS-LUA-Script-Menu/page19_

carloscampos775, 2017-06-21:
> 2793214-Modify-Betaflight-PID-VTX-Settings-w-TARANIS-LUA-
> Script-Menu/page19_

I'm using a X4R-SB. Going to try the firmware update anyway

carloscampos775, 2017-07-18:
Updated the RX firmware to current. Still the same issue

RipperDrone, 2017-07-30:
which version is your 'current fw'? There are different 'current' versions around for download...

carloscampos775, 2017-07-31:
FW# 170619

RipperDrone, 2017-07-31:
Have you enabled any additional sensors like GPS which might be misfiring or not be present at all?

closed 2018-04-21 by DieHertz