betaflight/betaflight #3310: Betaflight Logo

wolverin3, 2017-06-18:
This is just for aesthetic improvement but needed nevertheless. I would like to suggest that the Betaflight logo has a background so it is visible although during bright background. It should also display the version number of running BF and the target in use. Go or No Go?

Arakon, 2017-06-18:
Version number is shown on power up on the OSD, and can also be checked (even with the git commit number) in the OSD settings/FC info.

wolverin3, 2017-06-18:
Yes, just watched my saved video and found that it is there all along. I'm not sure if I'm the only one having trouble seeing the logo and those fonts, maybe I should try the Betaflight fonts rather than the Default. The scaling seems to be very rough too on my Quanum Cyclops V2. Both camera and goggles are on PAL.

mikeller, 2017-06-18:
@wolverin3: It's possible that the font memory on your OSD chip is corrupted. Try uploading the font from the configurator again.