betaflight/betaflight #3312: commit dc69c038869975df5aefc7ffc1f6c98edb48eae2

adrianmiriuta, 2017-06-18:
FrSky smartport telemetry not working

jflyper, 2017-06-18:
dc69c03 is #3307. Can you change the title to "#3307 breaks FrSky SmartPort" or something similar so that people know what the issue is all about?

mikeller, 2017-06-18:
@adrianmiriuta: What exactly is broken? And what makes you think that it is #3307? There is nothing in this patch that seems to be affecting SmartPort.

adrianmiriuta, 2017-06-19:
sorry the title was misleading, I do not know witch patch causes that.

mikeller, 2017-06-19:
Ah. Can you also add some information like how your FrSky RX is connected, and a config `diff`?

adrianmiriuta, 2017-06-19:
it is a test quad, hardware and configuration is ok. (worked with a 2 weeks old build)
Unless there are changes in serial configuration.
diff
--------------------------------------------------------------------
2017-06-19 @ 07:25:54 -- Running - OS: Linux, Chrome: 58.0.3029.110, Configurator: 3.1.1
2017-06-19 @ 07:25:56 -- Serial port successfully opened with ID: 1
2017-06-19 @ 07:25:56 -- MultiWii API version: 1.36.0
2017-06-19 @ 07:25:56 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-06-19 @ 07:25:56 -- Running firmware released on: Jun 18 2017 18:10:22
2017-06-19 @ 07:25:56 -- Board: LUX, version: 0
2017-06-19 @ 07:25:56 -- Unique device ID: 0x4200285633571120333131
2017-06-19 @ 07:25:56 -- Craft name: boogieL
2017-06-19 @ 07:25:57 -- CLI mode detected
2017-06-19 @ 07:39:20 -- Serial port successfully closed
2017-06-19 @ 07:39:29 -- Loaded release information from GitHub.
2017-06-19 @ 07:39:35 -- USB device successfully opened with ID: 1
2017-06-19 @ 07:39:35 -- Detected device with total flash size 256 kiB
2017-06-19 @ 07:39:36 -- Read protection not active
2017-06-19 @ 07:39:41 -- Erased 202 kB of flash successfully
2017-06-19 @ 07:39:51 -- USB device successfully closed
2017-06-19 @ 07:42:15 -- Serial port successfully opened with ID: 2
2017-06-19 @ 07:42:15 -- MultiWii API version: 1.36.0
2017-06-19 @ 07:42:15 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-06-19 @ 07:42:15 -- Running firmware released on: Jun 19 2017 07:38:53
2017-06-19 @ 07:42:15 -- Board: LUX, version: 0
2017-06-19 @ 07:42:15 -- Unique device ID: 0x4200285633571120333131
2017-06-19 @ 07:42:15 -- Craft name: boogieL
2017-06-19 @ 07:42:16 -- CLI mode detected
Setup
Ports
Configuration
Power & Battery
PID Tuning
Receiver
Modes
Motors
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.
$M>lc
CLI

# diff
# Betaflight / LUX_RACE 3.2.0 Jun 19 2017 / 07:38:53 (a907f4b)
name boogieL
servo 0 1000 2000 1500 100 90
servo 1 1000 2000 1500 100 90
servo 2 1000 2000 1500 100 90
servo 3 1000 2000 1500 100 90
servo 4 1000 2000 1500 100 90
servo 5 1000 2000 1500 100 90
servo 6 1000 2000 1500 100 90
servo 7 1000 2000 1500 100 90

feature VBAT
feature CURRENT_METER
feature UNUSED
feature AIRMODE
map RETA1234
serial 0 64 115200 57600 0 115200
serial 1 128 115200 57600 0 250000
serial 2 32 115200 57600 0 115200
led 0 0,0::ABIW:0
led 1 1,0::ABIW:0
led 2 2,0::ABIW:0
led 3 3,0::ABIW:0
led 4 4,0::ABIW:0
led 5 5,0::ABIW:0
led 6 6,0::ABIW:0
led 7 7,0::ABIW:0
mode_color 0 0 6
mode_color 6 1 2
mode_color 7 0 2
aux 0 0 1 1950 2075
aux 1 1 0 1550 1650
aux 2 2 0 1700 1800
aux 3 6 2 1950 2075
aux 4 7 3 1950 2075
aux 5 26 1 1950 2100
rxfail 6 (
rxfail 7 a
set gyro_sync_denom = 2
set blackbox_rate_denom = 4
set dshot_idle_value = 220
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 500
set bat_capacity = 1500
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 33
set current_meter = ADC
set ibata_scale = 237
set servo_lowpass_hz = 400
set deadband = 5
set yaw_deadband = 17
set tlm_inverted = ON
set frsky_unit = METRIC
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 1
set setpoint_relax_ratio = 20
set p_pitch = 30
set i_pitch = 35
set d_pitch = 10
set p_roll = 30
set i_roll = 35
set d_roll = 10
set p_yaw = 80
set i_yaw = 50
set d_level = 100
rateprofile 0

set thr_mid = 27
set thr_expo = 71
--------------------------

mikeller, 2017-06-19:
Try `set tlm_inverted = off`.

adrianmiriuta, 2017-06-19:
sorry,
it's again some changes in configuration  !
Configuring by hand, after full erase works !

closed 2017-06-19 by adrianmiriuta

jflyper, 2017-06-19:
@mikeller Bisecting tells me it's #3194 ...

mikeller, 2017-06-19:
@jflyper: N.b.:

`set tlm_inverted = ON`

`set tlm_inversion = ON`

Not the same, intentionally:

`tlm_inversion`: signal inversion is on;
`tlm_inverted`: signal is inverted *from the protocol default*.

jflyper, 2017-06-21:
OMG!

mikeller, 2017-06-22:
@jflyper: What's this for? :-)

slash4x4, 2017-11-15:
I am having the same problem updated to 3.2.2 on my Lux V2 flight controller and no longer get telemetry I have my xsr receiver hooked up to uart 4 . With s.port wire going to uart 5. On the Lux V2 at has its own dedicated a s.port pad on uart5 . Everything works fine on betaflight 3.1.6 after updating to 3.2.2 I no longer get Telemetry had to revert back to 3.1.6.

slash4x4, 2018-05-29:
> Did you find out what the issue was and how to correct?

No I never did