betaflight/betaflight #3317: Air-mode specific Minimum Throttle

amivit, 2017-06-19:
Why is there only a global minimum throttle? And not some configurable for different modes? 

Is it not true in order for air-mode to work properly, I must configure my ESC min-throttle to some place where the rotors spin, such as 1040. 

Problem is, I really dislike my rotors spinning when I arm (before enabling Air).

But because I need to have Air in rate/acro, I am forced into having. Does that make any sense? Is that really necessary? Or can I not have a separate min-throttle for air mode only... Hmm

mikeller, 2017-06-19:
I think what you want is `MOTOR_STOP`.

Cheredanine, 2017-06-19:
Motorstop would be overruled by airmode, you would also need airmode on a switch and not turn it on until you are in the air

amivit, 2017-06-19:
Oh! Motor stop would be a solution for me, I think. I switch between acro/rate+air for flying and angle for landing (without air). So I just need motorstop for my angle mode

mikeller, 2017-06-19:
@Cheredanine; I understood @amivit to say that he had air mode on a switch anyway:

> Problem is, I really dislike my rotors spinning when I arm (before enabling Air).

Cheredanine, 2017-06-19:
@mikeller 
Yeah I didn't follow the inference but looks like you are right

amivit, 2017-06-19:
Motor stop isn't what I want. I want my motors to immediately stop spinning in angle mode, when I give 0 throttle with my controller, without ruining my acro rate+air-mode by lowing my minimum-trottle. Makes sense?

mikeller, 2017-06-19:
That's easy. Enable `MOTOR_STOP`, put air mode on the same switch as angle mode. Air mode will dominate when enabled in acro mode, `MOTOR_STOP` will stop motors when air mode is disabled in angle mode.

amivit, 2017-06-19:
But doesn't `MOTOR_STOP` only work after the set delay and force me to rearm if I give 0 throttle?

Cheredanine, 2017-06-19:
Nope, it does what you just asked for, sends min command to motors at zero throttle provided air mode is not enabled.
This is all in the wiki

amivit, 2017-06-19:
I couldn't quite find any relevant information on the wiki regarding the specifics to `MOTOR_STOP`, however you are correct that it does give me the functionality I was aiming for. I just put the Disarm motors delay to 60, as that feature doesn't matter to me.

But now I can safely cut throttle in angle mode and know for sure the propellers aren't moving. But still have my 1040 min_throttle for Air-mode. Just the way I like

closed 2017-06-19 by amivit