betaflight/betaflight #3322: REVOLT V1 BETAFLIGHT 3.1.7 BLACK BOX

lucky13hc, 2017-06-20:
In betaflight 3.1.7 cannot read the onboard chip. In raceflight firms it works. Maybe a bug thanks in advance.

mikeller, 2017-06-21:
Hi @lucky13hc. GitHub is for discussion and questions around development of Betaflight. For questions about proper configuration / use of Betaflight, please see RCGroups: https://www.rcgroups.com/forums/showthread.php?2828750-Raceflight-Revolt-with-Betaflight-3-1

lucky13hc, 2017-06-21:
its not a configuration problem its a firm problem when i cannot see the on board data flash

mikeller, 2017-06-21:
I suspect you cannot see the data flash because your configuration is wrong. In the two months since 3.1.7 has been released, there have not been any other reports of this being broken, so that seems to be the more likely explanation. I don't think any of the Betaflight core developers have got this board, so it will be difficult to test this. How about you go to RCGroups, and ask fellow board owners about how they have configured their dataflash? If it turns out that it does not work for them either, we can always reopen this.

closed 2017-06-21 by mikeller