betaflight/betaflight #3324: [Feature Request] Implementation of VTX Setup through spektrum radios

sylarswe, 2017-06-20:
Hi!

I recent updates of the AIRWARE updates there has been added a VTX Setup menu where you can set channel and power output(similar to the LUA scripts for the taranis)
I have been doing some reverse engineering on this and it seems to work on all satellites .
Are there any software development currently on this functionality ?
If somebody is interested I can write down and post what I have on this

mikeller, 2017-06-21:
Hi @sylarswe. I don't think there is ongoing development on more features for the Spektrum protocols, but there certainly would be interest in getting this added. So it would be great if you could share your findings, and even better if you wanted to try your hand at adding support for your idea to Betaflight - we are certainly happy to help you along the way with this.

sylarswe, 2017-06-21:
Hi @mikeller 

Sounds great that there is interest in this!, I would love to get this working.
The VTX commands are sent through the regular satellites and are received as regular serial data on the satellite side with the same frame/protocol as the stick data.
Here is spektrums own interface technical data sheet on the receiver data stream
https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf

As you can see the channel identifiers (bits 14-11) are ranging from 0 to 11 in the regular stick data.
When you send a VTX setup command it replaces 2 channels (CH 0 Throttle and CH 6 Aux 2) with 2 commands on channel 12( effectively adding a 13th channel ).

The two commands have the same channel ID but give each a separate command data. 
The first word sets the channel according to the channel sheet I have linked down below. 
This seems to be a simple HEX value adding 1hex for every channel since 0x00 is immersion RC channel 1 and 0x20 is raceband 1, If there would be some variable/equation for a frequency there would be some similarities between the hex values that are on the approx same frequency.

The second word is output power where 0x08=25mW, 0x09=250mW and 0x0A=500mW, and if you set Pitmode active it just adds 0x10 to the value 0x18=25mW, 0x19=250mW and 0x1A=500mW.

Here you have some spreadsheets that I made from the data that I collected from my DX7 with all the available channel commands and some breakdown of the data stream
https://drive.google.com/drive/folders/0B9t3o9Pc-d9UQ0JWX3VpUTB6Nkk?usp=sharing

jflyper, 2017-06-21:
IIRC, there was a PR which included forged MSP message for handling VTX from Spektrum...

sylarswe, 2017-06-21:
Interesting!

Looks like this function is already implemented in that 3.0.2 branch.
I sat down and had a look at the code and I see that the whole MSP stuff has been moved since the 3.0.2 branch for the vortex 150 compared to the current 3.2.0 master branch so there is quite some work there just to see where all the MSP has been moved, I saw that there were some in the fc_msp.c and also a new folder named msp

closed 2018-04-21 by DieHertz