betaflight/betaflight #3329: SmartAudio not working on UART3 (Omnibus F4 targets)

digaus, 2017-06-22:
A month ago I had an Omnibus F4 with a Unify Pro HV Race and a Frsky X4R-SB, there I noticed that SmartAudio is not working when connected to UART3 (no data at all). If I connect SmartAudio to UART6 and SmartPort to UART3 everything is working.

Yesterday I got a new micro F4 which uses the Omnibus F4 target. This seems to have the same issue... SmartAudio not working on UART6 but on UART3... Odd thing is that SmartPort is always working, even on UART3.

jflyper, 2017-06-23:
@digaus 
Yeah, Unify is sometimes picky that it does not talk to a particular hardware UART on particular board.

So, your symptom is as follows

OMNIBUS F4 : UART3 x, UART6 o
a new micro F4 : UART3 o, UART6 x

Correct?

digaus, 2017-06-23:
No it works on both only on UART6. So at least it is consistent.

jflyper, 2017-06-23:
Oh, okay... What are the identity of these boards? Product URL would be the best.

digaus, 2017-06-23:
https://www.banggood.com/Betaflight-F4-Flight-Controller-STM32-F405-MCU-Integrated-OSD-p-1103940.html

https://www.banggood.com/Micro-20x20mm-Betaflight-STM32F4-F4-Brushless-Flight-Controller-Integrated-with-BEC-OSD-p-1132495.html

jflyper, 2017-06-23:
@digaus Oh... Basically, it is up to the user's responsibility for making something work on clone boards.

That being said,

(1) For OMNIBUS F4, I will test again when I get back to my bench, with non-PRO Unify.

(2) For micro version, I can't even find UART3 nor UART6 on the boards. Where are they?

digaus, 2017-06-23:
@jflyper 
Yep I know, I had to solder directly to the chip ;) bridged TX and RX so it holds better.

[image]

jumatic, 2017-06-27:
Its the Omnibus F4 Nano.
Use the omnibus exotic firmware for smartaudio softserial. Also tried to get it work on uart 6 and uart 1 with 3.1.5/6/7.
With firmware from exotic page i get the TBS Race work on TX1 and my DSMX Receiver on RX6. On the non-exotic version DMX doesnt work on RX6. This thsi was my Solution.

Saxin, 2017-07-26:
I'm experiencing somehow similar behavior on x2 Omnibus F4 V3 which were bought directly from myairbot (the designer and manufacturer). https://store.myairbot.com/flight-controller/omnibus-f3-f4/omnibusf4v3.html

Tried 3.1.6 , 3.1.7 and 3.2

Build #1: Uart6 - SBUS, Uart1 - Smart Port, Uart3 - Tramp HV (V1.26) telemetry.
Build #2: Uart6 - SBUS, Uart3 - Smart Port, Uart1 - Tramp HV (V1.26) telemetry.

In both occasions I can't get the VTX control to work - both Lua script and OSD.
Rewired build #2 and remapped PWM 5/6 for soft serial - works perfectly so the issue isn't in the VTX, I also have 4 of them (Tramp's) - all of them behave the same.

digaus, 2017-07-26:
@Saxin 
Normally SBUS should be on UART1 since it has the inverter for it. 
UART1 -> SBUS
UART3 -> SmartPort (inverted hack)
UART6 -> Tramp HV

You can also swap Tramp HV and SmartPort. Worked perfectly on all UARTS

Saxin, 2017-07-26:
This is not a pro - this is Omnibus F4 V3.
Take a look in the wiki [Omnibus SD](https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4)
Please edit your post to avoid confusion.

digaus, 2017-07-26:
Already did... so did you get the uninverted SmartPort signal?

Saxin, 2017-07-26:
Telemetry is working perfectly, yes XSR + **uninverted** signal.

Saxin, 2017-07-26:
@nathantsoi - I will appriciate if you can take a look at this issue, maybe it's some definition if the target file as UARTS changed a bit for OB V3 and up.

thanks.

jflyper, 2017-07-26:
@Saxin 
SmartAudio (Unify 5G8 PRO) is working on OMNIBUS F4 V3 / UART3.

OMNIBUS F4 V3
Betaflight / OMNIBUSF4SD 3.2.0 Jul 24 2017 / 23:40:49 (35f35fd94)
UART3_TX SmartAudio
UART6_RX SBUS

```
# diff
# version
# Betaflight / OMNIBUSF4SD 3.2.0 Jul 24 2017 / 23:40:49 (35f35fd94)¥
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
set serialrx_provider = SBUS
```

I can't think of any changes that made F4V3 incompatible with SmartAudio. It must be on your particular  hardware and setup. Is your setup new, or it has been working some time in the past?

Saxin, 2017-07-26:
I will test also TBS unify, I'm also in the dark here , it's 2 setups , new setups, both with same FC and tramp hv , I've tried both uart 3 and 1 on both .... as mentioned today morning tried with soft serial and it works perfectly.

Saxin, 2017-08-02:
@jflyper 

Cross checked - TBS Unify pro works, while Tramp HV doesn't.
Related to #3692

xeniter, 2019-02-19:
Had exactly same issue, in the betaflight wiki stand that they changed with Omnibus F4 V3 the serial Uart from 1 to 6. I assume they use inverter for it, so you not use anything bidirectional with this pin anymore. My solution was using uart1 for tramp, and any other bidirectional stuff on uart6. See [https://x.cmos.at/?p=238](https://x.cmos.at/?p=238)