betaflight/betaflight #3332: SD Card not working

alphat5, 2017-06-22:
Hi everybody. I Just finish a new fresh build with a Betaflight FC F3. Very haapy with this FC ;). But I need to have a blackbox log. So I install the SD card. But Betaflight doesn't recognise my SD "Fatal error". I tested with a NTFS and FAT32 ;).  If you can please help me out, it would be cool ^_^. 

Here is some stuff that maybe will help you to help me :p.

sd_info
SD card: Startup failed

dump
Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A00
resource MOTOR 8 A02
resource MOTOR 9 A08
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
[... 419 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

McGiverGim, 2017-06-22:
I see in your dump `feature -SDCARD` it seems the sdcard is deactivated.

Try in CLI `feature SDCARD`, then `save` to see if it works.

alphat5, 2017-06-22:
Yep, it worked, thanks. Didn't watched corretly ;). Thanks @McGiverGim  :).

closed 2017-06-22 by alphat5

Nellos-FPV, 2018-01-15:
I have de same problem but my FC is F4 Pro and my SDcard doesn't work i put the feature SDCARD on CLI but say INVALID COMMAND, help me please tks.

eried, 2018-07-03:
@Nellos-FPV did you solved that issue? I have the same problem. I am using OMNIBUSF4SD 3.3 but there is no SDCARD on the configuration to enable

Nellos-FPV, 2018-07-04:
Try flashering your controller to see if it will work, okay?

eried, 2018-07-04:
Yeah, I tried flashing all versions and below 3.2 (3.1.7) I can get the old sd to work (512 MB) but none of the newer fw work. I think is related with https://github.com/betaflight/betaflight/issues/5303