betaflight/betaflight #3337: 3.2.0 vbat not working

SkylionFPV, 2017-06-23:
I'm using an omnibus f4 v3 from ready to fly quads I updated the firmware to 3.2.0 from 3.1.7 to use dshot 1200, everything works the same as it did accept the voltage monitoring I just have this and it won't let me do anything with it, 
[image]
I saw someone else (ghost) had a similar problem back in march and I read the thread but I didn't really understand how he fixed it if anyone could clue me in I'd appreciate it

mikeller, 2017-06-23:
In order to configure battery monitoring for 3.2.0, you will have to use CLI, or use the latest configurator from the GitHub repository.

closed 2017-06-23 by mikeller

SkylionFPV, 2017-06-23:
how do I use the cli? I don't understand

mikeller, 2017-06-23:
https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

SkylionFPV, 2017-06-23:
I have no idea what I'm looking for here
I've always done everything through the main part of the configurator sp the cli is like alien language for the most part.
what things in the cli do I actually need to change?
and is 3.1.1 the newest version of the configurator?

SkylionFPV, 2017-06-23:
figured out configurator update thanks