betaflight/betaflight #3340: Safety Feature

hhoweson, 2017-06-23:
As an experiment I powered up the motors in the motor tab and then disconnected using the gui and unplugged the USB. The motors continued to spin, I feel that in this scenario the motors should be stopped straight away to avoid any idiots going full throttle with props, cutting the usb or yanking it out and having a spinning machine of death flying around with no way to stop it.

DieHertz, 2017-06-23:
What will keep idiots from arming with props and going full throttle?

hhoweson, 2017-06-23:
Nothing will XD I'm suggesting that in order to minimise damage motor power should be turned off when the flight controller becomes disconnected from betaflight

mikeller, 2017-06-25:
Unfortunately, out of the 100+ flight controller models currently supported by Betaflight, only 1 or 2 even have the hardware to detect when the USB connection to the computer is lost. In this respect, this feature would be useless for 98% of all users.

martinbudden, 2017-06-25:
Closing, since no practical solution. Note also motor tab clearly states props should be removed.

closed 2017-06-25 by martinbudden

ledvinap, 2017-06-25:
It is possible to use 'watchdog' fed by msp set motors command + modify configurator to send motor commands periodically. But it is probably not worth the effort

mikeller, 2017-06-25:
> It is possible to use 'watchdog' fed by msp set motors command + modify configurator to send motor commands periodically. But it is probably not worth the effort

I am not sure that this would not interfere with other use cases, where a user wants to set the motors to spin, and then use a different application to read data from the flight controller via MSP. And, as @martinbudden said, if a user is dumb enough to ignore a warning, and actively confirm that they have removed the props, despite the fact that they haven't, they probably can't be helped with a technological solution anyway.