betaflight/betaflight #3343: Allow Blackbox to overwrite oldest data, to always have latest flight data available

amivit, 2017-06-24:
A QoL enhancement, if there was a setting to always ensure blackbox would record latest data. When my 8MB flash is filled up, logging ceases. Why not just keep logging, overwriting the oldest data?

DieHertz, 2017-06-24:
Flash cannot be rewritten easily, you have to erase it first. Erase is performed on a per page basis and is a time-consuming operation, hard to implement it in realtime.
It may be possible to implement circular writes having some provisioning space, but this will decrease the amount of usable capacity even further.

amivit, 2017-06-24:
I see, that is certainly a dilemma. For now, I will personally just try and remember to download & wipe after sessions. But will leave this open for a contributor to either close, or look further into

martinbudden, 2017-06-25:
Closing, since there is no practical solution.

closed 2017-06-25 by martinbudden

lebed2045, 2022-05-30:
why just don't make an option to wipe it out before the flight? I'm happy to wait another couple of second every flight boot but have reliable blackbox data, because it's extremely impractical to connect to betaflight configurator every time before single flight. Blackbox is the most useful after the crash to see what really happen, so it's only last flight data that matters.

daleckystepan, 2022-05-30:
You can set BB erase on a switch to erase easily.