betaflight/betaflight #3347: BF3.1.7, BluejayF4, enabling SDCard in config delays PWM, stops APD Nano 22 ESCs from starting

ctzsnooze, 2017-06-25:
Hi everyone

I have some Bluejay F4 boards running betaflight 3.1.7, blackbox logging enabled, SDCard support checked,  SD card DMA access enabled.  

With Racerstar 30A ESCs, everything works fine, 5 slow beeps on the ESCs then they spin on arming.  All PWM protocols behave normally..

With APD Nano 22A ESCs, which have closed source firmware, this above configuration results in 6 fast then 3 fast beeps, and ESCs that will not spin the motors, regardless of protocol.

Disabling SDcard support solves the issue and the APD 22a ESC's start up normally - but then I cannot log to the card!

It seems that there is a delay in getting PWM for motors to the ESCs, so these particular APC ESCs then go to bootloader mode.  I am awaiting firmware to test, but to date have none.

I am wondering if there is something specific about enabling SD card support that would result in a significant delay activating motor PWM pulses?  Otherwise this issue may start affecting other ESCs.

I'm hoping the APD ESCs will get revised firmware that waits a bit longer for PWM pulses.

mikeller, 2017-06-25:
Interesting observation. Without looking at the code in detail, I would assume that SD card does some initialisation during the board initialisation, which will be adding some delay. If this delays the start of PWM signals for too much for these ESCs to initialise, then this is probably more of a problem of the ESC and not of the firmware. which is something to raise with the manufacturer.

ctzsnooze, 2017-06-25:
That makes sense.

The ESC manufacturer is happy to make a longer delay before the ESC 'gives up' on receiving a valid PWM signal, but of course many of them are already sold.

However it might be better if whatever SD card initialisation was required would not delay the start of the motor signals as much as it (apparently) does.

If someone who understood the code could maybe allow PWM to the motors to go ahead independent of SD card initialisation, that would be great!

mikeller, 2017-06-25:
I think what you are proposing is the wrong approach. From a safety point of view, initialising the ESCs before all of the components that are required for operation are properly initialised seems to be wrong. There are other on board components, like OSD / VTX / SPI RX require delays for initialisation too, all of them potentially derailing the initialisation of this particular type of ESC. The general approach taken by almost all ESC manufacturers is that timing of initialisation is not critical relative to when they are powered up, which is probably what this manufacturer should do too.

ctzsnooze, 2017-06-25:
The SD card is just a logging device, not a control device.

mikeller, 2017-06-25:
> There are other on board components, like OSD / VTX / SPI RX require delays for initialisation too, all of them potentially derailing the initialisation of this particular type of ESC.

As I said, the right approach is for the ESC manufacturer to change their firmware so that it does not rely on a fixed timing after power up. Making a requirement that an ESC can only be armed for a certain period of time after power up sounds like poor design in any case. There might be a bunch of valid reasons not to want to arm an ESC before certain other things have completed.

ctzsnooze, 2017-06-25:
I am not concerned about arming.

The safest approach is for min_command to be sent continuously, as soon as the FC is capable of doing so. 

There is no reason for the FC to not do this.

Arming can be delayed until all required systems are operational, I have no problem with that, but that is not the issue I am reporting.  

The issue that concerns me is that the ESC should be receiving min_command as soon as possible to ensure keep all ESCs from spinning, but it is delayed considerably if you have to wait for the SD card to enable, or other things to happen.  

In any case the manufacturer has modified the firmware to increase the delay, and now it works.  But the better solution is for the FC to start sending min_command as soon as practical to ensure the ESC's don't respond to noise or glitches or enter programming modes inappropriately.  This was a problem in the past and is why PWM at min_command normally starts immediately.  

The SD_Card iniitialisation code should not delay starting PWM at min_command.

blckmn, 2017-06-25:
@ctzsnooze the ESC should receive a series of valid min_command pulses before it considers the pulse valid and then honour any further increase in throttle (and start the motors spinning).

At the moment as the FC boots there is absolutely no signal on the wire, which is the right thing to do. The ESC should never start spinning a motor in this case.

It also shouldn't need to have protection on timings on no valid signal - it could conceivably wait 30 seconds or an hour. It is the transition from no signal, to a series of min_command pulses, that offers the protection.

Although you are correct in the case of SD card (it has no bearing on control), it is still a very real issue for the ESCs, as an examples is we wouldn't be able to implement soft reboot for such things as system crash recovery, or clock speed changes if the ESC has these timing requirements for valid signals - and if not met goes into a "do nothing" state.

mikeller, 2017-06-26:
@ctzsnooze:

> The issue that concerns me is that the ESC should be receiving min_command as soon as possible to ensure keep all ESCs from spinning, but it is delayed considerably if you have to wait for the SD card to enable, or other things to happen.

I think you do not understand how arming for ESC works on PWM / OneShot / MultiShot:

Any correct ESC design requires a certain minimum duration of a min_command signal to be received before the ESC goes from 'unarmed' to 'armed'. When armed, if the signal goes outside of the range of min_command to max_command for a certain duration, the ESC will drop back to 'unarmed'. While in the 'unarmed' state, the ESC will not spin the motors, no matter what it gets as an input signal.

This means that the safe thing to do is to **not** send min_command until everything is ready and initialised, and thus keep the ESCs in unarmed state. Once armed by receiving a min_command signal, they will be sensitive to spurious signals.

ctzsnooze, 2017-06-26:
OK thanks for that explanation.

I didn't realise that there was a separate 'armed' state in an ESC, separate as in independently of the flight controller 'armed' state (in which, following a switch or other 'arming' action, the flight controller sends min_throttle to the ESCs and becomes responsive to throttle stick inputs).  

What you say makes sense.

Do you know if betaflight itself waits for the SD card to initialise also (ie before allowing the FC to arm)?

blckmn, 2017-06-26:
@ctzsnooze yes Betaflight waits until the end of the init phase (where everything is initialised, including led strip etc) before allowing arming :)

borisbstyle, 2017-06-26:
This indeed sounds like a bad implementation on the ESC. 
The timer between powering up the ESC and sending the minimum signal length to even start processing ESC signals seems too low on these. 
I think this is failsafe timer on the ESCs?

ctzsnooze, 2017-06-27:
Yep OK I agree, the ESC should not have that time dependency.  

However, honestly, I'm not sure we need to wait for an SD card to initialise before being able to fly.  I'd prefer the shortest possible boot to fly time; most likely on a racing quad I would not care if the SD card was not ready, especially if logging is not intended.  

Maybe if blackbox is not active, we could ignore SD card startup - why delay in that case?

mikeller, 2017-06-27:
That's what I tried to explain to you above: The initialisation sequence is quite complex, and there are a lot of constraints on what is initialised when. I am not sure that we are in a position to be able to read the configuration to determine if blackbox is enabled or not at the point that we are initialising the SPI bus for the SD card - probably not.

If you prefer fast startup to the ability to write blackbox logs, feel free to build your own firmware with SD card / blackbox support disabled.

closed 2017-06-27 by mikeller

ctzsnooze, 2017-06-27:
Ha ha I can already just disable SD card, and it starts fast.  On a racing quad, nothing else causes a meaningful start delay, only SD card support.

mikeller, 2017-06-27:
Ah, I thought you'd already tried that with no success. In general, disabling anything that you don't need will improve performance.

ctzsnooze, 2017-06-27:
Disabling it solved the problem, but of course then I couldn't record.  So I could either fly, or record, but not both  :-)

ledvinap, 2017-06-27:
@ctzsnooze : Removing SD card instead of disabling SD does not work?

blckmn, 2017-06-27:
@ledvinap The problem does not occur if it is disabled. It is a combination of enabling the SD card - and the time it takes to initialise this (yes its relatively short, but long enough to cause a problem) and the particular ESCs in use. 

Those ESCs will go into a failsafe - do nothing - mode if there is not a valid signal within a very short period of time. This is actually resolved with @ctzsnooze getting new firmware on the ESCs.

But those ESCs are in market, and with out the update the timing on the boot sequence on the FC until signal to ESCs is too long. 

This could be resolved by moving the SD card initialisation until after motor outputs are initialised (and valid signals started), BUT must still occur before ARMing is permitted (as BB is triggered on ARMing).

It is also resolved by the ESC manufacturer updating the firmware - which is a much better cause of action - as it is not expected behaviour.

ledvinap, 2017-06-27:
The idea was not inserting SD card instead of disabling it ...

My opinion is that ESC should be fixed. Moving SD initialization will complicate code with little gain.

blckmn, 2017-06-27:
Yes we definitely agree the ESC firmware should be fixed.

Though the idea of removing the SD card - and logging to it at the same time are mutually exclusive. The OP wanted to use the ESCs and BB logging to SD card.

mikeller, 2017-06-27:
> Though the idea of removing the SD card - and logging to it at the same time are mutually exclusive. The OP wanted to use the ESCs and BB logging to SD card.

I thought at least on boards with SD card detect wired up the insertion of a card would be detected and handled even after initialisation? (Even then, it's a clunky workaround.)