betaflight/betaflight #3357: Current Meter Type isn't working

Makkeysn, 2017-06-26:
When VBAT and CURRENT_METER is activated and "onboard ADC" selected as Current Meter Type, Betaflight is rebooting....after the reboot, the "Current Meter Type"-box is empty/unchecked.
With BF 3.1.7 it's working well!

mikeller, 2017-06-26:
You'll have to get the latest version of the configurator from GitHub in order for the battery configuration to work with latest unreleased 3.2. Or else configure the battery settings in CLI.

mikeller, 2017-06-26:
3.1.1. is the latest released version of the configurator. It works with all released versions of the firmware. If you are using unreleased versions of the firmware, then using the unreleased version of the configurator as well is the thing that is most likely to work.

AndersHoglund, 2017-06-26:
One problem is that both the released version and the current development HEAD version says: `Configurator: 3.1.1` Only way to tell the difference is by the installation path. Why not have the same versioning strategy as BF, bump development version when a releases is made.

iH8win0ws, 2017-06-27:
Now it says `Configurator: 3.1.2`.

mikeller, 2017-06-27:
Yes, the version was bumped, and now you know you're using the development version. :-)