betaflight/betaflight #3358: BF 3.2.0 - dshot throttle problem after arming

TheIsotopes, 2017-06-26:
since [jenkins build 1379](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1379/) and the following i have throttle probs after arming ... the motors start sporadically rotate (feels like at mid throttle) and when i disarming, the motors still rotates ... only accu plug off stops the motors

```
# Betaflight / REVOLT 3.2.0 Jun 25 2017 / 10:33:19 (b77b866)

# name
name HX200

# feature
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200

# led
led 0 10,10::AW:0
led 1 9,10::AW:0
led 2 8,10::AW:0
led 3 7,10::AW:0
led 4 6,10::AW:0
led 5 5,10::AW:0
led 6 9,5::CW:6
led 7 8,5::CW:6
led 8 7,5::CW:6
led 9 6,5::CW:6
led 10 6,2::CW:6
led 11 7,2::CW:6
led 12 8,2::CW:6
led 13 9,2::CW:6

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1600 2000
aux 1 13 1 1000 1200
aux 2 15 1 1000 1600
aux 3 26 0 1600 1800
aux 4 31 0 1200 1400

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set min_check = 1000
set serialrx_provider = SUMD
set blackbox_rate_denom = 8
set blackbox_device = SPIFLASH
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set current_meter = ADC
set battery_meter = ADC
set use_vbat_alerts = OFF
set use_cbat_alerts = ON
set vbat_scale = 109
set ibata_scale = 236
set pid_process_denom = 1

# profile
set dterm_notch_hz = 0
set pid_at_min_throttle = OFF
set p_pitch = 66
set d_pitch = 30
set p_roll = 56
set d_roll = 24
set p_yaw = 80

# rateprofile
set rc_rate = 110
set rc_rate_yaw = 110
set roll_srate = 80
set pitch_srate = 80
```

mikeller, 2017-06-26:
Thanks for reporting, @TheIsotopes, looking into this now.

alfag4, 2017-06-26:
Hello, same problem on OmnibusF4 dshot

TheIsotopes, 2017-06-26:
thx @mikeller dshot works again ... closed now

closed 2017-06-26 by TheIsotopes