betaflight/betaflight #3361: 3.1.7 punchout - rolls and yaws to the right

playlet, 2017-06-26:
Hi!

I've been flying my Wizard (SPRF3) on 3.1.7 since it came out but this little problem makes me a little uncomfortable to keep using it.
On every punchout (no airmode), my quad dips to the right (roll and yaw).

I know that this will be fixed in the next release but that is a loooooong time ahead.

I love the way my quad flies on 3.1.7. It is so much better than 3.1.6 that I'm using now to avoid this problem. Airmode is a no go for me, I don't want to destroy my escs in a crash.

Would you mind releasing 3.1.8 (3.1.7 + fix for that)? October is a long way ahead.

Best regards, Ivan from Croatia

mikeller, 2017-06-27:
Hi @playlet.

There have been discussions about releasing the next version earlier than the announced October release date. But since there has been so much change since 3.1.7, releasing another version based on 3.1.7 would take an immense amount of work, and testing, so this is probably not going to happen, it's gonna be a release with all of the new features.

You could always try the latest build of 3.2.0 from here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/

AndersHoglund, 2017-06-27:
Releasing a 3.1.8 based on 3.1.7 should of course not include any of the massive changes in 3.2.0 on the master branch.  Such a maint release should be branched out  from the 3.1.7 release tag and only the required bug fixes should be added. A merge back to master could then also be done, if applicable.

playlet, 2017-06-27:
I know that there are a lot of people still on 3.1.6 or older fw because of this. If this were to be fixed and published, all of us could enjoy the 3.1.7 fw. It was so much better than 3.1.6. I don't know why, it just was. :)

mikeller, 2017-06-27:
@AndersHoglund: Are you willing to spend the 40 or so hours to backport the 20 or so fixes into pre - parameter group firmware? Thanks.

AndersHoglund, 2017-06-30:
No, that is not the way to do maintenance. 
From what I understand there is only one line in one file that needs changing. Branch out "patch_3.1.8" from the "v3.1.7" tag (or the HEAD of patch_3.1.7 branch) and fix this single thing in the way of a Cactus-ish model. This will be a branch on a branch, as we do not have a generic "maint_3.1.x" branch for continuous maint.  Not quite the Cactus way, but git has no problem with multi branching.

mikeller, 2017-07-02:
@AndersHoglund: Why do you think that this particular bugfix is so much more important than the 20 or so bugfixes for bugs in 3.1.7 that have been added to 3.2.0, so that it merits being put into a release of its own, without fixes for the other known and fixed bugs?

In general, I am opposed to doing releases that exclude fixes for known and fixed bugs, unless it's to fix a bug that has serious safety implications. And I don't see that this has safety implications. One of the reasons being that everybody who is affected by any of these bugs will open a new issue, (rightfully) asking 'why has this not been fixed'?

AndersHoglund, 2017-07-02:
OK. if they are real end user problems then they should/could/might be taken in to a 3.1.8 maint release, one by one in separate PRs. It is a priority thing, a decision to be made issue by issue and in what order. We should really have a condensed list of known issues on every release.  I can start 3.1.7 issues:

1. Punchout
2. DYSF4 Sbus inverter.
3. ..... 20 <Fill me in>

We do not need to release fixes for everything at one time, kind of old "big bang" thinking that would only delay problem fixing. Or nothing at all will be fixed, as the current situation is. Take to top <X> issues and release 3.1.8. Then if there is a cry for more fixes, and no workarounds are available, do 3.1.9. Then 3.1.10, 3.1.11 etc etc. Unless 3.2.0 already is out by then.

AndersHoglund, 2017-07-02:
And really, we are getting OT now. This is a thread for one issue, that has a probable fix. It could be a candidate for a 3.1.8 release. The other 20 or so issues should be discussed elsewhere.

mikeller, 2017-07-02:
You are right, let's just stop wasting time on discussing or preparing half-assed fixes.

closed 2017-07-02 by mikeller

RipperDrone, 2017-07-02:
I know this topic is closed, tried to search in which issue report it is still marked to be fixed in some future BF version, but could not find it anywhere else.

Just wanted to confirm this issue on my Omnibus F7 build / DShot600, Nightly 3.2.0 from June. Quad does a right/front dip after punching or cutting throttle abrubtly. Cannot be fixed by higher I values (tried up to 50)...

nathantsoi, 2017-07-02:
With airmode on?

RipperDrone, 2017-07-02:
Yes, AM was On. But no anti_gravity yet...