betaflight/betaflight #3364: FrSky receiver firmware flashing

andrejpodzimek, 2017-06-27:
It would be awesome to flash FrSky receiver firmware directly from the FC, via SmartPort and/or SBUS.
 (This may apply to some other receiver types as well.) The process could look (from the user’s perspective) similar to e.g. BLHeli firmware updates. 

FrSky receivers can be [flashed using OpenTX](https://youtu.be/eYVk2lWPJxU) via a cable connected to the Taranis transmitter. This is inconvenient for multiple reasons.
1. Tiny receivers, [such as the XM series](http://www.frsky-rc.com/download/view.php?sort=Firmware&down=380&file=Firmware-XM%20PLUS), don’t have a plug connector and require desoldering and resoldering, twice per firmware upgrade. That comes with a lot of (thermal) wear and tear on such a small component.
2. Not everyone uses a Taranis. For me a Devo 10 with a 4-in-1 module is a much better option, because it speaks way more protocols than a Taranis, without external hardware or further adjustments.

Some might argue that RX updates are infrequent. However, whenever I travel to the USA and take my FrSky drones with me, the possibility to change the RX firmware (and the protocol variant) from LBT to FCC and back for conformance would be great. Devo 10 with Deviation supports both LBT and FCC out-of-the-box, but the RX is the main issue, because I don’t want to do too much soldering e.g. on tiny drones like Storm Loki-X2 / Loki-X16.

The receiver firmware flashing functionality [is implemented in OpenTX](https://github.com/opentx/opentx/blob/eb7c73668f55026c57b880027acc77f1bd2ee00a/radio/src/io/frsky_sport.cpp#L301). Would it be technically feasible to have such a feature in Betaflight?

DieHertz, 2017-06-27:
Try using CLI `serialpassthrough` functionality to forward USB-serial requests to your receiver flashing port, and just use FrSky-recommended PC flashing tool.
I don't see a necessity for some FrSky flashing-specific code in BF if serial passthrough works.

andrejpodzimek, 2017-06-27:
Thanks for the hint; I didn’t know about the `serialpassthrough` feature. I can’t use [the FrSky recommended tool](http://www.frsky-rc.com/download/view.php?sort=tool&down=374&file=Tool-FrSky%20update%20sport), because it won’t run on any of my systems. FrSky doesn’t seem to provide the source code of these tools. I stumbled upon [this tool](http://openrcforums.com/forum/viewtopic.php?f=52&t=2914#p41724), but it looks somewhat outdated, with no “official” repository. Not sure if it’s safe to use that with a recent FrSky receiver. The last resort option may be to pass the `usbserial` device into a VirtualBox container with a suitable spyware running inside it, but that sounds like too many moving parts.

DieHertz, 2017-06-27:
Use a VM or give Wine a try. There's virtually no chance that BF will have proprietary code for flashing 3rd party hardware, so it's up to FrSky to make their tools open-source, cross-platform, or both.

ledvinap, 2017-06-27:
@andrejpodzimek : Be careful when flashing FrSky products. At least in past firmware was encrypted but not verified before write, so it was quite easy to brick receiver ...

andrejpodzimek, 2017-06-27:
I don’t think OpenTX has proprietary flashing software in it. The implementation is open-source, IIUC. It’s just that the serial protocol is somewhat hard to grasp from the source code (otherwise I would have already tried to just hack the flashing command and data transfer in C) and it doesn’t seem to be available in the form of a standalone flashing tool. [This looks promising](https://github.com/fishpepper/OpenSky) and has some pointers to flashing tools, but it seems to only support a small subclass of devices and there’s no sign of support for the 16-channel FrSky protocols (called `FrskyX` in Deviation). Apart from the FCC versus LBT motivation, I also have at least two XM+ receivers with an ugly RSSI-related bug — they lose signal in mid air at random, usually once per flight, for a second or so. Pretty annoying. That’s why I’m looking for upgrade options, preferably soldering-free and Taranis-free. Will try Wine, to see if that works with `serialpassthrough`. Well, if I brick the receiver, then I’ll have to do the soldering. That’s life.

andrejpodzimek, 2017-06-27:
Anyway, I'll close this, because it looks like Betaflight has all the needed functionality (`serialpassthrough`). The missing part is a BLHeli-like interface for FrSky receiver flashing. Will try to contact FrSky, but I my guess would be that it won't help much.

DieHertz, 2017-06-27:
Give me more info and I'll see what can be done, integrating user-experience stuff is always a good thing if it can be done with little effort. :-)

andrejpodzimek, 2017-06-27:
[This](https://github.com/opentx/opentx/blob/eb7c73668f55026c57b880027acc77f1bd2ee00a/radio/src/io/frsky_sport.cpp#L301) appears to be the high-level view of the flashing procedure. [This code](https://github.com/opentx/opentx/blob/eb7c73668f55026c57b880027acc77f1bd2ee00a/radio/src/io/frsky_sport.cpp#L24) seems to have some relevant protocol definitions. The packet[...] structures in the functions below show what the protocol messages look like. The protocol includes [some checksumming / parity done in `sportWritePacket()`](https://github.com/opentx/opentx/blob/eb7c73668f55026c57b880027acc77f1bd2ee00a/radio/src/io/frsky_sport.cpp#L158) and that's probably all that's done with the data before it's written out [using DMA](https://github.com/opentx/opentx/blob/eb7c73668f55026c57b880027acc77f1bd2ee00a/radio/src/targets/taranis/telemetry_driver.cpp#L92). The [entry point from the GUI/menu is here,](https://github.com/opentx/opentx/blob/656daa75fafa47803b42c45eaa08068c96628441/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L154) I think; it's the Flash External Device feature.

What confuses me though are the `sport` (SmartPort) identifiers everywhere. The small XM+ receiver doesn't have anything like SmartPort telemetry (which is usually on bigger receivers with a 4-wire connector (+, -, SBUS to the FC and SmartPort from the FC)). The XM+ has just 3 wires, +, - and SBUS. Yet in spite of that, the Taranis can flash a simple 3-wire receiver (as shown in [the video](https://youtu.be/eYVk2lWPJxU) already linked above). I'm not familiar with these low-level protocols at all, so I don't feel bold enough to just hack my own `sportFlashDevice()` and give it a try with `serialpassthrough`. But it would be awesome if that could work!

mikeller, 2017-06-27:
@andrejpodzimek: Just for completeness' sake: ersky9x and 9XR Pro (and probably other 9X derivatives) and derived TX have the functionality to flash FrSky firmware too.
But I agree that the capability to do so via flight controller and `serialpassthrough` is much more convenient. There is a user on the Betaflight slack (http://www.betaflight.ch/) that recently reported having used `serialpassthrough` successfully to flash a FrSky Rx.

ledvinap, 2017-06-27:
@mikeller : Flashing capability may be implemented in configurator, if someone has a lot of free time ;-)

From firmware perspective, serialpassthrough-over-msp may be easily implemented (just packetize raw serial data)..

mikeller, 2017-06-27:
@ledvinap: Configurator is definitely preferrable over using precious flash space to do it in firmware. But even better would be a standalone 'FrSky configurator' a la blheli configurator.

ledvinap, 2017-06-27:
Flashing is pretty trivial - Smartport format is documented, commands are simple.
It should be few tens of lines in javascript ...

DieHertz, 2017-06-27:
Sure I'll try to minimize flash space if I'm going to implement it

JamesCordell, 2017-06-27:
Hi 

@andrejpodzimek 

Just to inform you that there is an opensource Frsky firmware flasher tool here.

https://github.com/JamesCordell/frsky_updateV2

I used it successfully on Ubuntu to flash my XSR receiver with LBT firmware.

andrejpodzimek, 2017-06-27:
@JamesCordell Thanks a lot for this bit of information! For some reason I just didn't find your repository, although I had been searching for FrSky update tools quite thoroguhly. BTW, as for *“to be used with the fuc-3 usb lead”*, is it feasible to simply use Betaflight and `serialpassthrough` instead of the USB lead, with the obvious advantage of not having to desolder the receiver?

andrejpodzimek, 2017-06-28:
@JamesCordell Is there by any chance a command-line variant of the tool? I installed [this package](https://aur.archlinux.org/packages/qextserialport/) to tackle the first build error I encountered, but it turns out that it no longer supports Qt4 in any shape or form. Well, I might have an ancient VM from the Qt4 era somewhere, in the worst case. (Attempts at http://github.com/KDAB/Qt4to5 didn't work either; it's been 5 years and `clang::tooling` has changed too much.)

JamesCordell, 2017-06-28:
HI

Yes on newer builds you need to install apt-get install qt4-default on Ubuntu 16 for example. I have no experience of serialpassthrough yet maybe ill need it one day.

Arakon, 2017-06-28:
Serialpassthrough doesn't appear to work for this. The frsky tool doesn't even see the serial port of the FC in the first place, likely due to the fact that it's a VCP.

mikeller, 2017-06-28:
@Arakon: I had a quick look, and on linux the tool seems to be adding /dev/ttyACM0 to the port list, so this part works.
The one thing to consider though is that, for flashing to work, the flashing seems to have to start immediately after the RX is powered up. So if your RX is powered from the flight controller, and gets power as soon as USB is connected, this won't work, since you will have to go into the CLI first, and configure the serialpassthrough.

@JamesCordell: There is a typo in your `README.md`: `apt-get install qu4-default`

JamesCordell, 2017-06-28:
Thanks for point out the type o. I remember that the xsr had to be plugged in at a critical point, this was a pain. This could make serialpassthrough a no go as now the power line has to be part of the protocol. We will probably see a opensource replacement for the receiver before anyone can be bothered to fix this stuff. There are more important bugs and features to be swatting in Bf before we get to this OMHO.

mikeller, 2017-06-28:
@JamesCordell: Well, an easy way to address the power up issue is to wire the RX so that it gets power from the PDB. Then it is the same as flashing ESCs: Get everything ready, then plug in a battery. Your flashing tool could just keep sending the first packet until it gets an answer from the RX. The flashing tool built into ersky9x already does it this way: You start the flashing process, and after that you plug in the RX.

> We will probably see a opensource replacement for the receiver

Yes we will. ;-) But even then, for the large number of users out there that own XSRs, being able to flash without having to fiddle with plugs would be a very welocome thing.

MJ666, 2017-06-28:
Could be also handle like the bind mode for the Spektrum satellites. Here you also need to disconnect power and reconnect again.

mikeller, 2017-06-28:
> Could be also handle like the bind mode for the Spektrum satellites. Here you also need to disconnect power and reconnect again.

Not sure if this will work, will depend on the timing requirements for the firmware update, which might be too narrow to allow for Betaflight to initialise to a point where it can read the config and set up serial passthrough.

brycedjohnson, 2017-06-29:
I flashed the XSR with serialpassthrough and the windows frsky tool.  There was an issue on a rmdo were I needed to set msp to 57600. #3288 

VCP with omnibus worked well.

The biggest problem as mikeller mentioned is that reset is required to hop into the bootloader there appears to be only a brief window to get in there.

andrejpodzimek, 2017-07-08:
@JamesCordell The need to use Ubuntu 16 (or any Ubuntu at all) doesn't look like a solution. (Qt4 will eventually disappear also from Ubuntu anyway.) Up-to-date distros (e.g. Fedora, Arch, Tumbleweed) can no longer build Qt4 stuff (easily). I thought I'd try to extract a simple command-line tool [from here](https://github.com/JamesCordell/frsky_updateV2/blob/master/frskyupdate.cpp), but it's too intertwined with the Qt4 API and libraries (`QextSerialPort`). There should be a way to do the flashing just in POSIX calls (`open()`, `ioctl()`, `tc{get,set}attr()`, `read()`, `poll()`, `write()`), portable and independent of GUI toolkits. A first step that crossed my mind would be to run the existing tool under `strace` to see what `QextSerialPort` does in terms of system calls in this particular scenario. But I'm not sure how many receivers I'd brick before getting it right, so I guess I'll just give up on this idea.

DerDominik, 2017-08-19:
@brycedjohnson
Did you configured something special in serialpassthrough to flash via windows frsky tool?
I'm trying on a SPRacingF3 and S.Port is connected to softserial. But in frsky tool it hangs in "finding device".
I tried with rx powering before and after switching to serialpassthrough.

Any suggestions?

mikeller, 2017-08-19:
@derdominik: I think you need to power on the RX **after** starting the FrSky tool.

DerDominik, 2017-08-19:
Yes, I tried also with powering the rx after choosing the file. In the tool the green led is blinking and it says "finding device"

brycedjohnson, 2017-08-19:
For spracingf3 I had to set the msp to 57600 for some reason as well.  Remember to connect back to betaflight at 57600 and set it back when you are done.

DerDominik, 2017-08-19:
I tried also with msp set to 57600, same problem. Maybe softserial is the problem as I can use max. 19200?

mikeller, 2017-08-19:
@DerDominik: The new softserial implementation should be able to handle 57600, otherwise it would not work for SmartPort in the first place. But not sure if it can handle 57600 and a high data rate (i.e. firmware image being pushed), SmartPort itself is only short frames and a lot of idle time in between. Maybe @jflyper can help here?

TimTimNL, 2017-09-01:
@brycedjohnson i tried this for hours to connect the smartport with serial passthrough BF would set up the serial connection but the frsky tool just never saw the XSR i tried various timings of when to boot the xsr.
Anything im mising here ?

jflyper, 2017-09-01:
@TimTimNL Which port are you using for SmartPort?

TimTimNL, 2017-09-01:
@jflyper UART1 so my command reads `serialpassthrough 0 57600` which returns port 1 already open intializing passthrough.

TimTimNL, 2017-09-01:
Just tried BF 3.1.7 for my KAKUTEF4 as i saw a ton of people complaining of smartport issues.
With 3.1.7 after the serialpassthrough command it spits out a ton of characters.
From there i just minimize BF disconnect the XSR from its plug. 
Start the FrSky tool and load the file so the tool is waiting for device.
And plug it back in but it never sees the xsr.

tried this with manually setting the MSP to 56700 aswell same results.

jflyper, 2017-09-02:
@TimTimNL You should disconnect, not minimize the BFC.

TimTimNL, 2017-09-02:
@jflyper I've tried that aswell. No go.

gloob, 2017-10-29:
Tested it in macOS with a Windows 10 vagrant box, it works really great, as commented by @TimTimNL you need to set the MSP speed to 57600 of the port too. It's really great to flash your receivers without soldering them from the FC.

Arakon, 2017-10-29:
One thing I am wondering about.. when flashing through the Taranis, the receiver gets sent into flashing/BL mode by the transmitter. So there should actually be a command that causes the receiver to reset, without having to unplug and replug. If this could be triggered by BF as part of the passthrough command (i.e. "serialpassthrough 1 57600 flash"), that would make it much easier.. however, I am not sure if that command/trigger has been researched and found.

mikeller, 2017-10-29:
@Arakon: The boot loader is triggered by the Taranis switching on the power supply to the RX, and then sending some form of init sequence within a pre-defined timeout. This is the usual way serial bootloaders are implemented on ATMega MCUs. (See also the instructions for flashing with 9XR Pro / er9x, which require the user to start the flash programming, **and then** connect the Vin pin of the RX, because the 9XR Pro does not have switchable power pins in the extension bay.)

Arakon, 2017-10-29:
Ah, too bad.. I had hoped there was a reboot sequence that could be sent.

mikeller, 2017-10-29:
@arakon: Not really. Imagine this reboot sequence being sent accidentally,  by a bug in the telemetry code, while you are hovering over that pond...

DerDominik, 2017-11-21:
Didn't tested a firmware update of the RX with actual Betaflight versions, but yesterday I saw the new FrSky Firmware with FPort Protocol.

Maybe someone knows, if there's an easier/better solution for a firmware update via flightcontroller implemented? Or maybe it is possible to make a suggestion to FrSky for this function in newer RX firmware versions with FPort Protocol?

mikeller, 2017-11-21:
@DerDominik: Yes there is. As described in https://github.com/betaflight/betaflight/issues/3364#issuecomment-312121695, you can flash the RX with `serialpassthrough`. The only thing you need to do is connect the RX Vin to a voltage source that is not supplied by USB, so that it is not on when USB is connected but no battery. Then power up the flight controller by connecting USB, initialise `serialpassthrough <smartport port>`, exit the configurator, start the FrSky update tool, *and then* connect a battery to power up your RX.

thepiper92, 2017-12-04:
I want to get RSSI on my Kakute with an XM+. Can I use serialpassthrough on BF 3.2? From reading, luckily my XM+ is only powered when I use battery, and not through usb. What do I have to type in CLI? Just serialpassthrough? Where do I get this FrSky program?

mikeller, 2017-12-05:
@thepiper92: Yes you can use serialpassthrough. The command is `serialpassthrough <port number>` with the port number being 1 less than the UART number shown in the ports tab (i.e. `0` for UART1, and so on). You can get the FrSky tool from the FrSky website, where you can also get the required firmware update.

thepiper92, 2017-12-05:
Awesome! Thank you! So Uart 3 is 2. I just type 2, not <2> I gather?

mikeller, 2017-12-05:
@thepiper92: Correct.

zielonydomek, 2018-01-20:
Hi ! Have you success ?

thepiper92, 2018-01-20:
It did not. Just ended up using tx to flash

spencerschumann, 2018-03-25:
I also tried and failed to update an XM receiver using `serialpassthrough`. I don't have access to an Frsky transmitter for the update, and I also don't have the Frsky "STK" cable, but I do have an FTDI USB to serial breakout board for flashing Arduino boards.

I created a tool to flash receivers using these FTDI boards on Linux. It's a wrapper around OpenTX's flashing algorithms. It's at https://github.com/spencerschumann/opentx, in case anyone else might find it useful.

mikeller, 2018-03-25:
@spencerschumann: Thanks for that, this might be helpful for some users.

Just for completeness' sake: ersky9x also has a function to flash RX from the JR module connector.

rhys100, 2018-03-25:
Hi Guys, i successfully flashed using windows after a bit of muckaround.

I made a blog on correct steps - https://www.phaserfpv.com.au/blogs/blog/flash-frsky-receivers-via-your-flight-controller-using-smart-port-betaflight-serialpassthrough

Posted cheat notes below

Download FRSKY Tool or "Tool-FrSky update sport" from here - https://www.frsky-rc.com/frusb-3-fuc-3/
Download the firmware and note the file you will flash. EU has LBT in the filename, international has FCC in the filename
Plug in your quad via USB and goto ports, and note the SMART PORT Uart you are using. You will use one number less in next code line. EG. UART 6 type serialpassthrough 5
Then goto CLI and type serialpassthrough 5 57600
Disconnect from Betaflight Configurator 
Unplug your receiver now so it has no power what so ever
Open up FRSKY tool and select the file
Now plug in your quad via LIPO
Now plug in your Receiver.
If successful it should now have current firmware version and hardware version number in the FRSKY tool.
Click Download and be patient :)

MJ666, 2018-03-30:
I tried this with an PikoBLX clone (Flying Egg) and an XM RX and an PikoF4 and an XM+ RX without success. Both have BF 3.3.0 installed. @rhys100 what FC you have used for flashing? For the PikoF4 may be the inverter is interfering? Also for flashing it looks to be an half duplex serial protocol is needed. There may be some specific settings required for the serial port?

mikeller, 2018-03-30:
@MJ666: My experience from doing this a number of times is that, if the inverter / settings work for SmartPort, then flashing over the same UART will work (and `serialpassthrough` just keeps the same serial port settings if the port is already in use). In the case of the XM / XM+ this will be less useful, since they don't already are connected through a bidirectional inverter for SmartPort. But setting up SmartPort on the UART that is normally used for SBus, and then invoking `serialpassthrough` might still help. (Obviously not for F4, since the SBus inverter normally isn't bidirectional.)

MJ666, 2018-03-31:
Also tried with SPort telemetry enabled on the serial RX port for the F3. This setting does not stick? I also tried with "set serialrx_halfduplex=on" but this does also not work. I may need to use an FTDI and find an diode somewhere to build an flash tool for my XM/XM+ receivers. I need to enable the RSSI on some of them.

mikeller, 2018-03-31:
As I said, I never tried on XM/XM+, it's possible that their baud rate / bit count is different from SmartPort's for flashing. You could also use the JR module pins on your TX.

MJ666, 2018-04-01:
@mikeller i dont own an FrSky TX and using a deviated Devo10 TX. No way to flash the RX with this one. I have a few FDTI adapters and only need to find a diode to build a adapter. The FrSky STK is quite expensive.

mikeller, 2018-04-01:
Agreed, FTDI is the best option then. No point buying a fancy adapter for a one-off.

Arakon, 2018-04-01:
You can also use a resistor (2.2k for example) instead of the diode.. in fact, my FTDI only worked with a resistor, diode didn't do the trick.

[14 more replies]

closed 2018-05-13 by stale[bot]