betaflight/betaflight #3366: BF 3.2 LED aux adjustment

bug5532, 2017-06-27:
In BF 3.1 when you set an aux channel to adjust LED colors, 1000 would be the color you set, and then it would cycle through all colors, until you were back to where you started at 2000.
in BF 3.2 1500 is the color you set, but you only seem to be able to get half the colors going to 2000/1000 not the full spectrum.

mikeller, 2017-06-27:
The way how adjustable LED colours are configured has changed (and you can now adjust more than just the hue). See https://github.com/betaflight/betaflight/pull/3151#issuecomment-304440403 for an explanation of how it's done.

bug5532, 2017-06-27:
Nice, cheers man

closed 2017-06-27 by bug5532

GareFPV, 2017-08-09:
The new adjustable LED colours dont work well with a rotary switch. I want to change the colour with my Taranis for showracing and dont dont want to configurate the LED colour every race with the GUI/Laptop. Is there another solution i didn't know?