betaflight/betaflight #3367: OMNIBUSF4SD RSSI

papadkostas, 2017-06-27:
Can't get it to work. The input RSSI signal is width modulated as seen on the scope.
I've found another person which experiencing the same problem.
https://www.reddit.com/r/Multicopter/comments/6fok9j/rssi_on_an_omnibus_f4_pro_v3_no_dice/

jumatic, 2017-06-27:
Analog or PWM?
For PWM you need to get the firmware at omnibus exotic. For Analog you need to build your own or i can maybe send you.
Have done it for my Omnibus F3 and F4 Pro V2, but current sensor is then disabled.

jflyper, 2017-06-27:
@papadkostas @jumatic RSSI ADC was left out on OMNIBUS F4 by accident. Let me see if it works and if it does, I'll submit a PR.

papadkostas, 2017-06-28:
_First of all my board is Omnibus F4 PRO V3_

@jumatic It's PWM. I've tried already exotic build but either way it didn't worked even if i followed the instructions, by disabling RSSI ADC and disable RSSI channel from the receiver tab.
I've tried RSSI pad and PWM 5 & 6 with softpwm, but still didn't worked.
Why current sensor gets ? since PWM needs timer and current sensor an ADC?

@jflyper Can you try exotic build as well for Direct FrSky RSSI input (Auto Range version) ? 
please let me know if you need any testing.

jflyper, 2017-06-28:
@papadkostas I'm the author of the "Direct FrSky RSSI input (Auto Range version)" and also the "Exotic OMNIBUS". ...

papadkostas, 2017-06-28:
So does it work for you? I''m curious..

jumatic, 2017-06-28:
have build my own firmware with analog rssi adc.
@ target.h
#define USE_ADC
//#define CURRENT_METER_ADC_PIN   PC1
#define RSSI_ADC_PIN            PA0
#define VBAT_ADC_PIN            PC2

And yes it works.
 
Will we get a second adc with next updates?
 
Also have tried the pwm, but i think its cause by my analog signal, i only 
get 0 and 100. So i have done it on my own.

 
 
 

 
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/betaflight/betaflight"}},"updates":{"snippets":[{"icon":"PERSON","message":"@jflyper 
in #3367: @papadkostas @jumatic RSSI ADC was left out on OMNIBUS F4 by 
PR."}],"action":{"name":"View 
Issue","url":"https://github.com/betaflight/betaflight/issues/3367#issuecomment-311518525"}}}

mikeller, 2017-06-28:
> Will we get a second adc with next updates?

I don't think there is an effort going on for that. In reality it shouldn't be needed, since one ADC can handle up to 16 inputs.

jflyper, 2017-06-29:
@jumatic I think you don't need to disable `CURRENT_METER_ADC_PIN`; all three channels can be used at the same time.

> Also have tried the pwm, but i think its cause by my analog signal, i only
get 0 and 100. So i have done it on my own.

The software-PWM RSSI input is designed to capture PWM signal _edges_. It is pointless to use it with voltage level encoded RSSI signal.

jflyper, 2017-06-29:
I will submit the target.h with RSSI_ADC_PIN enabled anyway.

papadkostas, 2017-06-29:
So for PWM RSSI we'll need to filter the signal to get analog value?
Won't be possible to get it straight into?

jflyper, 2017-06-29:
@papadkostas 
> So for PWM RSSI we'll need to filter the signal to get analog value?

ATM, yes.

> Won't be possible to get it straight into?

When #2292 is done, you can do that. ATM, it works for F4, but has some problem with F3.

papadkostas, 2017-06-29:
OK, so if the prolem with softPWM only remains to F3 would it work ok with F4SD target?
I will try to build in the afternoon.

jflyper, 2017-06-29:
@papadkostas 
Try it. I made a couple of small changes, so be sure to pick up the latest branch.

papadkostas, 2017-07-01:
Built latest branch, RSSI only works for analog now which previously didn't worked at all.
In my case, the value is constantly changes as i turn on and off the transmitter but it's not accurate which is what I expected since my signal is PWM. So I assume we'll have to wait for the #2292 as I'm not interesting on filtering the signal.

jflyper, 2017-07-01:
@papadkostas You can try #2292 also.

papadkostas, 2017-07-01:
@jflyper Ok!! I'll will try to figure out how to do so :D

papadkostas, 2017-07-03:
Sorry for the deyal, just built this https://github.com/papadkostas/betaflight/tree/RSSIsoftPWM
Voltage and current meter seems not to be working anymore.
RSSI signal is soldered directly to RSSI pad. RSSI value is not changing, just stays at 0%.
But on the debug screen i can see it working actually. Please have a look [here](http://oi63.tinypic.com/538c60.jpg). 
Am i doing something wrong with the build?

jflyper, 2017-07-03:
@papadkostas What is your receiver???

papadkostas, 2017-07-03:
@jflyper FRSky Delta 8

jflyper, 2017-07-03:
@papadkostas The signal format (the encoding) is not the one the rssi_softpwm can handle atm...

Can you come up with the spec? Cycle time and duty ratio for 0 and 100%?

papadkostas, 2017-07-04:
@jflyper the signal period is 5ms. The DC for no signal is about 5% and 95% for full signal.
I will get proper measurement with DSO, but i highly doubt that the measurements are off more than 5%.
What signal can be handled properly for soft pwm?

jflyper, 2017-07-04:
@papadkostas The current code is based on Frsky D and X series receivers, in which period is around  1ms and duty ratio changes from 2~98% (yes, the shortest pulse will be like 2us and the longest is 998.)

For this reason, the current code just discards measurements with cycle time larger than 2ms. I think I can just loosen up this constraint to little less than 10ms. (Can't make it more than 10ms = measurement cycle time.)

papadkostas, 2017-07-04:
@jflyper should i look into soft rssi source file? Im out atm..

jflyper, 2017-07-04:
I'll try to make a time to take a look at it tonight.

jflyper, 2017-07-05:
@papadkostas Checkout the latest commit. I've changed the magic number from 2000 to 10000. (Measurement cycle time was 100msec - 10Hz).

papadkostas, 2017-07-05:
@jflyper I've saw it yesterday while traveling, found it hard coded in the source file.
Will test this in the afternoon.

jflyper, 2017-10-23:
Aftermath: Omnibus F4 does not need pwm rssi, as it has a valid ADC for RSSI pin.

closed 2017-10-23 by jflyper