betaflight/betaflight #3371: Cc3d led strip

ijustwannafly, 2017-06-28:
Using st link v2 programmer and swd port cc3d board has been flashed with betaflight 3.1.7 (openpilot bl removed).

Led strip, configured as enabled, does not recognize leds (0 remaining).  Flashing bf 2.9 led strip works and shows 4 remaining.

I looked into using docker to try to build a cc3d build based on 3.1.7 with led strip but I have had no success.

Is there any way to get a 3.1.7 bf build for cc3d with led strip support?

I'm sure this board is looked at as old/eol but it still works and so I would like to run it in the ground before replacing.

Many thanks.

mikeller, 2017-06-28:
Try this: 

[betaflight_3.2.0_CC3D.hex.zip](https://github.com/betaflight/betaflight/files/1108543/betaflight_3.2.0_CC3D.hex.zip)

LED_STRIP seems to fit into CC3D, so I've enabled it in the target definition (for now): #3372.

ijustwannafly, 2017-06-28:
Many thanks!

I had to assign LED-STRIP 1 to resource B04 (previously motor 5).  Led shows 32 remaining though only 4 out of the 8 are being programmed.  I am working on figuring it out as all 8 are operational in 2.9.

Also, if you would be so kind, there used to be a bp6 build (beeper on pin 6).  I am hoping I just have to reassign the beeper 1 resource to the motor 6 pin though ai am not sure if there is anything else beyond that that is needed.

closed 2017-06-28 by martinbudden

mikeller, 2017-06-28:
@ijustwannafly: Thanks for your feedback. Beeper pin should indeed be assignable now. Please report back if you can't get the LED_STRIP to work properly.

ijustwannafly, 2017-06-28:
Mikeller,

Everything works properly including buzzer/beeper by assigning to previous motor 6 output.

I will definitely look into getting a docker build to compile for future use.

ijustwannafly, 2017-06-28:
I spoke too soon.

After configuring the leds and then attempting to reconfigure all my other settings...I am unable to calibrate my escs with the motor tab.  

I also cannot save my min, max throttle and min command values.  Changing the values to 1050, 2000, 1000 and then saving and reboot resets it to 0.  

Any possible way to use the 3.1.7 base with led strip support as opposed to the 3.2 you used?

Many thanks.

mikeller, 2017-06-28:
> Any possible way to use the 3.1.7 base with led strip support as opposed to the 3.2 you used?

Ok, this is awkward. I just tried to patch 3.1.7 to support LED_STRIP on CC3D, but it turns out it actually is enabled in 3.1.7 already (for non _OPBL). No idea why this doesn't work for you.

If you want to use 3.2.0, you'll have to get the latest unreleased configurator from GitHub to get a working UI.

ijustwannafly, 2017-06-29:
mikeller,

It was strange too for me.  When I flashed 3.1.7 with my st link v2 and swd port I verified that the opbl was removed...at least for my board gcs no longer recognizes it and I don't need to do the power trick to get betaflight configurator to recognize it.

Also, when I flashed 3.1.7 I wasn't able to map LED_STRIP 1 to any resource because bfconfigurator returned an error saying the name didn't exist.

I grabbed the 3.1.7 file from the releases page...the one that does not specify for opbl.

I will make a dump of my current config on this test build and reflash 3.1.7 and let you know.

ijustwannafly, 2017-06-29:
Ok so here's my latest.  I thought I was missing something so I redownloaded the 3.1.7 hex for cc3d non-opbl.

Dumps from 3.1.7 and 3.2.  Apparently the 3.1.7 does not list LED_STRIP as a resource.
[cc3d_3.2_test_dump.txt](https://github.com/betaflight/betaflight/files/1110593/cc3d_3.2_test_dump.txt)
[cc3d_3.1.7_dump.txt](https://github.com/betaflight/betaflight/files/1110592/cc3d_3.1.7_dump.txt)

mikeller, 2017-06-29:
Yes. Since the CC3D does not seem to have a dedicated pin for LED_STRIP, you will have to assign it to an unused pin.

ijustwannafly, 2017-06-29:
Mikeller,

Understandable.  I should reword it.  

On flashing 3.1.7 and enabling led strip it does not show up as an assignable resource as the dump file shows.

On flashing the test 3.2 build and enabling led strip it shows up as an assignable resource as the dump shows.

mikeller, 2017-06-29:
Ha, a closer look revealed that LED_STRIP has to be explicitly defined in 3.1.7. Try this:
[betaflight_3.1.7_CC3D.hex.zip](https://github.com/betaflight/betaflight/files/1110625/betaflight_3.1.7_CC3D.hex.zip)

ijustwannafly, 2017-06-29:
Ok next set of tests.

I flashed back and forth between the 3.2 test and the 3.1.7 test you posted to make sure I was getting repeatable results.

Flashed the 3.1.7 test and led strip does not work.  Assigning led strip 1 to resource b04 (motor 5) does not activate led support.  On the 3.1.7 test after enabling and reassigning led strip it does not show up in the resources or the resource list.

Flashed the 3.2 test and led strip does work.  Assigning led strip 1 to resource b04 (motor 5) activates led support.  On the 3.2 test enabling and reassigning led strip shows led strip 1 in the resource and resource list.

One difference that stands out between the two versions...on the 3.1.7 test build when enabling and reassigning led strip it does not show up anywhere.

On the 3.2 test when enabling and reassigning led strip it shows up in the resource and under resource list it is attached to DMA1 Channel 6.

It does not attach to the DMA1 channel on the 3.1.7 test.

mikeller, 2017-06-29:
Hmm, I don't think it makes sense to dwell on the past, so to speak. Building 3.1.7 with LED_STRIP support was a stab at getting something that works without spending much effort of it. If it doesn't work the way forward is to use 3.2.0.

ijustwannafly, 2017-06-29:
I have no problem using 3.2 especially since it actually works.  I flew my quad with the cc3d board and the 3.2 test you gave me.  I was just concerned that the gui wasn't reflecting certain values or allowing certain functions.

I guess for now until the official 3.2 comes out I can continue to use the 3.2 test on my cc3d since I can fly it.

Many thanks.

mikeller, 2017-06-29:
The GUI works of you use the latest unreleased configurator from GitHub.

ijustwannafly, 2017-06-29:
You got me there.

My configurator version reads 3.1.1 the same as the last release on github.  I can't seem to find the unreleased version you are referring to,.

mikeller, 2017-06-29:
You'll have to pull the latest unreleased version directly from the repository, see the instructions in `README.md`.

ijustwannafly, 2017-06-29:
mikeller,

Many thanks!  Configurator from the git repo does work as intended with the 3.2 test build.

PR #3372 (2017-06-28): Enabled LED_STRIP on CC3D, fixed includes.
Fixes #3371.