betaflight/betaflight #3386: [Feature request] Serial ESC driver for konze protocol

Cesco1, 2017-06-29:


mikeller, 2017-06-29:
Looks interesting. What ESC hardware does this run on?

Also, would you like to open a pull request for this change so we can work out the implementation details?

Cesco1, 2017-06-30:
The protocol is part of simonk firmware. It is enabled by default in afro-series esc, and can be enabled in most simonk firmwares. Condition is the rxd pin is connected to the pwm input cable. Esc is have flashed are hk blueseries, hk ubec series, afro series. The simonk firmware needs a little patch for the bootloader and we need to set the individual MOTOR-ID.
Description: https://www.rcgroups.com/forums/member.php?u=185255

I am worknig on a blheli-s patch, but im a noob in c51 so no sucess yet.

>would you like to open a pull request

Sorry im a noob with git and i dont understand the structure of serial port assignement in betaflight. I might need some help.

mikeller, 2017-06-30:
Interesting. I had assumed that you'd need a 4in1 so that there is a 1:1 serial connection, but I guess in this case it is wired up as a bus.

> Sorry im a noob with git and i dont understand the structure of serial port assignement in betaflight. I might need some help.

Then look at this as a chance to learn git and GitHub. Everybody in here is happy to help new contributors. As a first step, create your own fork of Betaflight, and apply your changes to a branch of it. Once you got that working, creating a pull request is easy.

Cesco1, 2017-06-30:
> I guess in this case it is wired up as a bus.

Yep, the yellow wire here: http://i.imgur.com/HM2soYy.jpg

mikeller, 2017-06-30:
I see. You are Swiss by any chance? ;-)

Cesco1, 2017-06-30:
The cofee cup doxes me right?

Cesco1, 2017-07-01:
yes