betaflight/betaflight #3387: Lux f4 osd

CC210, 2017-06-29:
Having trouble getting the battery voltage monitoring to save on esc or on board Lumineer f4 osd aio. Wraith 32

mikeller, 2017-06-29:
Can you please change the title to something that is more specific about what the problem is?

CC210, 2017-06-29:
I closed it just may be an issue with battery voltage monitoring saving on the lumineer f4. New to the site

closed 2017-06-29 by CC210

mikeller, 2017-06-29:
For next time, try to do what you are asked to do, and don't just close the issue. Even after closing the issue, it will stay around. If it has a nondescript title, it will waste the time of everybody who is searching GitHub for answers.

Also, it will probably work if you use the configurator from the GitHub repository.

CC210, 2017-06-29:
Well I couldn't change the title from where I an and really doesn't matter to me what you think

mikeller, 2017-06-29:
@CC210: This is not about me, this is about advancing a cause (making Betaflight better) as a community. If you are not willing to contribute, please stay away from GitHub in the future.

CC210, 2017-06-29:
Well I would like to make issues aware but the do what you ask comment was a bit overboard and the title I thought pointed to lux f4 programming/ possible error with an issue I was having saving the battery voltage on a lumineer f4 aio board that is brand new to the hobby so what would you like that title to read sir.

mikeller, 2017-06-29:
I don't think asking a user to change the title of an issue from simply being a board name to being a board name **and** a description of what is actually not working is overboard at all. Try putting yourself into the shoes of somebody having a problem with the same board, and trying to search for a solution on GitHub. If everybody created tickets with just the board name, they would potentially have to read through 100s of issues, instead of being able to only look at the ones that are actually related to their problem.

I've already changed the title.

Also, if you'd searched for something like 'battery configuration', you'd found #3357 as your first hit, together with a solution for your problem.

CC210, 2017-06-29:
I'll keep that in mind and thanks for pointing that out to me.