betaflight/betaflight #3388: BF 3.2.0 - sporadic motor twitch on oneshot

TheIsotopes, 2017-06-29:
since [jenkins build 1388](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1388/) i have sporadic motor twitch on disarmed fc ... with the build before is all good ... imho #3365 is the culprit ... @blckmn pls can you check this

here the diff from last good build:
```
# Betaflight / SPRACINGF3EVO 3.2.0 Jun 28 2017 / 11:11:12 (9616001)
name Trifecta
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource SERVO 1 B01
mixer TRI
feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
map TAER1234
serial 1 64 115200 57600 0 115200
led 0 4,2::AW:0
led 1 5,2::AW:0
led 2 6,2::AW:0
led 3 9,2::AW:0
led 4 10,2::AW:0
led 5 11,2::AW:0
mode_color 6 1 3
aux 0 0 0 1600 2000
aux 1 13 1 1000 1200
aux 2 15 1 1000 1600
aux 3 26 0 1600 1800
set align_gyro = CW270
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set align_acc = CW270
set acc_hardware = NONE
set min_check = 1000
set max_aux_channels = 2
set serialrx_provider = SUMD
set blackbox_rate_denom = 4
set min_throttle = 1055
set max_throttle = 1960
set current_meter = NONE
set vbat_scale = 108
set tri_unarmed_servo = OFF
set pid_process_denom = 2
set dterm_notch_hz = 0
set pid_at_min_throttle = OFF
set rc_rate = 110
set rc_rate_yaw = 110
set roll_srate = 80
set pitch_srate = 80

```

blckmn, 2017-06-30:
I'll check tonight and let you know.

blckmn, 2017-06-30:
Do you have any scope output you can share?

TheIsotopes, 2017-06-30:
What do you mean by that?

blckmn, 2017-06-30:
https://github.com/betaflight/betaflight/pull/3392

This might solve the problem.

TheIsotopes, 2017-06-30:
@blckmn thx, seems to be good again
using latest master with your fix #3392 ... throttle new calibrated and no more twitch on disarmed status

closed 2017-06-30 by TheIsotopes

jflyper, 2017-07-14:
Related #3490