betaflight/betaflight #3390: Erratic behavior in airmode

thepiper92, 2017-06-30:
I recently converted my Torrent 110 to a iPeaka 12a and Femto FC. All is well other than air mode. I have accelerometer turned off as I will never have use for it, but airmode is not working. When on (verified in betaflight with the arm switch showing highlighted) it idles up to an appropriate rpm. While holding the quad, once past around half throttle it randomly jumps to new full throttle, and it is a wavering throttle that leaves the quad shaking. The new iPeaka heated up too. I thought maybe it was due to holding the quad, but once I got it hovering it just rocketed into the ceiling (luckily no damage) and wouldnt stop until I disarmed. I tried again holding the quad and raised the throttle, to the point it acted odd, and decreased it. It took a good second for the throttle to even decrease. I have since tried antigravity and motor response is fine. What is the issue with my setup?

Please not I have motor 4 soldered to the led pad to allow dshot, and I have Dshot 600 active.

Cheredanine, 2017-06-30:
That has nothing to do with airmode not working, in fact it is air mode working very well, you have another issue possibly

soinds like you have some sort of oscillation occurring mid throttle, there is a nice section in the wiki faq on how to deal with that, or it could be your filtering, either way it isn't a software bug
Try the rcgroups thread on bf, you will get a better response there than on the software forum for this type of problem

thepiper92, 2017-06-30:
Thank you. Could it be pid values not being the same as they were on the stock FC? Where can I find this faq?

Cheredanine, 2017-06-30:
Seems unlikely, 
At the top of the page there is a link to the wiki, the FAQ is in that

closed 2017-07-02 by mikeller